zynthian/zynthian-ui #129: Timeout loading engine causes UI to freeze.

smiths73v3, 2019-06-07:
Timeout while loading engine causes UI to freeze with the "wave" animation continuing forever.
Need to login via ssh to restart zynthian UI
```
>systemctl restart zynthian
``` 

Here is the log capture of the start engine failure:
```
Jack: JackClient::ClientNotify ref = 5 name = ZynMidiRouter notify = 18
Jack: JackClient::ClientNotify ref = 5 name = ZynMidiRouter notify = 18
ERROR:root:Can't start engine ZynAddSubFX => Timeout exceeded.
<pexpect.pty_spawn.spawn object at 0x6df3c470>
command: /usr/local/bin/zynaddsubfx
args: ['/usr/local/bin/zynaddsubfx', '-O', 'jack-multi', '-I', 'jack', '-P', '6693', '-a', '-U']
buffer (last 100 chars): b'Open name = zynaddsubfx\r\nJack: JackClientSocket::Connect : addr.sun_path /dev/shm/jack_default_0_0\r\n'
before (last 100 chars): b'Open name = zynaddsubfx\r\nJack: JackClientSocket::Connect : addr.sun_path /dev/shm/jack_default_0_0\r\n'
after: <class 'pexpect.exceptions.TIMEOUT'>
match: None
match_index: None
exitstatus: None
flag_eof: False
pid: 10187
child_fd: 22
closed: False
timeout: 20
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile(b'\n\\[INFO] Main Loop...')
INFO:root:OSC target in port 6693
INFO:root:OSC server running in port 17170
DEBUG:root:ZCTRL 'Channel': 0 (0 -> 14), None, None
Jack: JackClient::ClientNotify ref = 4 name = jackpeak notify = 18
```

jofemodo, 2019-06-13:
Sorry, i don't know hot to reproduce this issue. I would need a more detailed description ...

BTW, can you reproduce this issue in the last nightly build?

- https://os.zynthian.org/2019-06-13-zynthianos-stretch-lite-1.0.0.zip

smiths73v3, 2019-06-22:
I have not been able to reproduce it recently. It was caused when adding multiple Synth engines sometimes a new engine would not load. The first error message seen was:
```
ERROR:root:Can't start engine ZynAddSubFX => Timeout exceeded.
```

riban-bw, 2019-06-22:
I was still seeing this occasionally a couple of weeks ago. I wonder whether recent changes to error handling may have improved this issue.

jypma, 2020-01-17:
I did upgrade a few days ago... I'll re-update and let you know the results, with detailed steps as well.

smiths73v3, 2020-06-03:
Have not seen this issue since last July 2019.

closed 2020-06-03 by smiths73v3