zynthian/zynthian-ui #130: Pianoteq is not working in Aruk RC2

Djihatch, 2019-06-10:
**Describe the bug**
When using a pianoteq layer. No sound comes out. Blue "m" appears when a note is played on a master USB keyboard but the Vu meter stays at 0. In fact, it acts exactly as if no layer were selected.
Everythings works ok with layers using synth or sf2 players...

**To Reproduce**
Steps to reproduce the behavior:
1. Go to new layer/pianoteq/ch#1/harp (27) (doesn't work with other instruments too)/daily
4. The layer is created but no sounds come out when usb keyboard pressed

**Expected behavior**
Sounds should have been played when master usb keyboard is used.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Setup:**
 - Aruk SD2 image
 - zynthian box V2 (without case and without midi in and out ribbon + connector)
 - Last software version I guess (image downloaded and software updated on the box menu).

**Additional context**
At least another user (riban) has the same problem. Hes expects that the demo version of pianoteq is out of date (https://discourse.zynthian.org/t/5-midi-devices-connected-only-one-is-working/3016/20)

riban-bw, 2019-06-10:
I can confirm Pianoteq demo is not working, no audio output.

jofemodo, 2019-06-11:
OK! I've updated the demo version to 6.5.1, the latest one, so it should last quite a bit befor it's deprecated.
The next "green" build will integrate the new version.

riban-bw, 2019-06-11:
Ha! I knew me downloading a nightly last night would prompt some change that would require a repeat.... Cheers!

jofemodo, 2019-06-11:
Running this lines in your today's image should do the job:

```
update_zynthian.sh
$ZYNTHIAN_RECIPE_DIR/install_pianoteq_demo.sh
reboot
```

riban-bw, 2019-06-11:
Thanks @jofemodo - that works a treat now. A side effect is that it _seems_ to xrun less often. I can still trigger an xrun if I really push it but with only Pianoteq loaded and most other features disabled, e.g. no network MIDI, OSC, etc. it is pretty good. From release notes:
> Improved performance on the ARM platform -- cpu requirements of version 6.5.1 are still higher than version 6.4.1, but they are lower than 6.5.0.
I wonder whether it is worth trying (or making an option) version 6.4.1 if that seems to be the most CP efficient version. I think 6.5.1 is better than 6.0.3 though.
@Djihatch please confirm this fixes things for you. (Version discussion is for another ticket.)

Djihatch, 2019-06-11:
I just come back from work. The download of the last version is in progress. I'll give you the feedback tomorrow I hope.
Thanks to be so quick and responsive.
I'll let you know.

Djihatch, 2019-06-11:
@riban-bw I'm unlucky... It's not working for me. I've downloaded latest image (2019-06-11-zynthianos-stretch-lite-1.0.0.zip, right ?). Used the "update software" from the GUI. And It just acts as before : synths and sf2 are ok, but no sounds come from pianoteq layer....

Djihatch, 2019-06-11:
Oops. It's OK now, i'm sorry.
I misunderstood : I thought that the lines given by jofemodo where only for riban because he didn't get the last image.
I ran them and everything's ok now !!!
Thanks.

jofemodo, 2019-06-12:
It should work out-the-box. I will check it ...

Djihatch, 2019-06-12:
Thank you. And thanks for this great project !

riban-bw, 2019-06-12:
I have had scenarios where Pianoteq doors not work if reloaded (as default layer) after a system reboot. This might be related to the system clock updating and influencing the trial period. Restarting Pianoteq generally fixes it.

Djihatch, 2019-06-12:
You may be right. I rebooted 2 or 3 times, but I may have counted the initial boot (I don't remember). Maybe one more boot should have done the trick.

jofemodo, 2019-06-12:
OK! Tomorrow's build should be fine out-the-box ... i hope ;-)

jofemodo, 2019-06-14:
OK! It's working in today's night build:

https://os.zynthian.org/2019-06-14-zynthianos-stretch-lite-1.0.0.zip

closed 2019-06-14 by jofemodo