zynthian/zynthian-ui #131: CUIA "ALL_OFF" restarts UI

riban-bw, 2019-06-10:
Sending `/cuia/all_off` via OSC restarts the UI.

**To Reproduce**
1. Open Python3 interpreter
2. Enter `import liblo`
3. Enter `liblo.send(('zynthian.local', 1370), "/cuia/all_off")`

**Expected behaviour**
Currently sounding notes are silenced.

**Actual behaviour**
User interface restarts

**SOFTWARE**
zyncoder: master (f672128) 
zynthian-ui: master (4a1e025) 
zynthian-sys: master (3db6429) 
zynthian-data: master (676c32a) 
zynthian-webconf: master (588c830) 

We had a similar issue in the past. Is this a regression?

jofemodo, 2019-06-12:
The problem in the past seems to not being totally solved.  I hope it's solved now.
See also: https://github.com/zynthian/zyncoder/commit/80862b6ed1aca412d710656040192d485c58db80

closed 2019-06-12 by jofemodo

jofemodo, 2019-06-12:
FYI, i've implemented note state tracking in the MIDI router. This open the door for new features and improvements, some of which have been already commented in other threads (i.e. zynthian-ui#66)

commit 5ca27387 (2019-06-12): Fix #131: "All Off" provokes a segmentation fault. See Zyncoder f672128.