zynthian/zynthian-ui #132: Mathematical manipulation of control data

riban-bw, 2019-06-13:
If I map Channel Pressure (aftertouch) to an engine parameter, e.g. LFO frequency, the range of control will invariably be inappropriate with initial zero pressure asserting too low control value and full pressure asserting too low or too high value to give the playing performance required.

It would be advantageous to allow some method of manipulating control data e.g. to allow minimum, maximum control value to map to different minimum, maximum signals feeding an engine parameter. This may be a simple min / max map or some mathematical formula.

Mididings may be able to provide a solution or maybe some extra configuration within zynmidirouter / midi profile.

jofemodo, 2020-06-17:
This can be partially addressed by using some MIDI-tool, taking profit of the new MIDI-chain feature.

jofemodo, 2020-10-07:
The MIDI FXs implementation solve partially this issue, although mididings could be a fine (advanced) tool to integrate.
Perhaps we could create a separate task for "Integrating Mididings" ...