zynthian/zynthian-ui #133: Mute on system reboot.

wyleu, 2019-06-13:
**Is your feature request related to a problem? Please describe.**
When a machine is rebooted from the webconf a considerable period of Digital Noise ( looping buffer) can be generated before the machine goes silent for the rebbot.

**Describe the solution you'd like**
Silence would be preferable, especially in a live scenario ( hifiberry Amp+ users :- ) ) 

**Describe alternatives you've considered**
Muting on an external mixer is the only option and if you are rebooting the odds are you are airly busy at the time already, so a purposely muted output would probably be what you would actually prefer to happen.
**Additional context**
Couldn't get a recording using the zynth,   cos.

jofemodo, 2019-06-13:
Disconnecting everything from jack's system output should do the work ... ;-)
I will do it ASAP!

jofemodo, 2019-06-13:
Hi @wyleu !

I just realized that this is already being addressed on #134 

Regards!

closed 2019-06-13 by jofemodo