zynthian/zynthian-ui #135: Map MIDI filter without channel fails

riban-bw, 2019-06-14:
If you add a midi filter with channel = ALL it may not parse, e.g.

MAP CC#114 => CH#15 NON#12

This should map CC 114 from any channel to Note on 12 on channel 15 but the web configuration fails. Setting a channel in the source makes it work, e.g.

MAP CH#0 CC#114 => CH#15 NON#12

The web interface offers the option for a source MIDI channel of "ANY" so one would expect this to work (and is what I want in this case). A work around is to add 16 rules, one for each channel.

I don't know whether this is an issue in the web config or zynthian but I experience it whilst configuring in the web conf so am reporting against this project.

jofemodo, 2019-06-14:
Exactly, the channel list on the left should have the same length than the one on the right.
@mheidt, until this limitation is removed on the rule parser, do you think it could be enforced by the webconf's rule builder?

mheidt, 2019-06-14:
Actually I don't want to program all rules in javascript in webconf.
That's what the parser is meant for.
But the detailed error message should be displayed, which isn't yet.

ANd besides, I think, this issue should be fixed  with the logic, that in case that the source has n channels, but the destination only one, the mapping is ok.
If it is n:m, the current error message is okay.

jofemodo, 2019-06-14:
OK! I will improve the parser so it works with "1=>n" and "n=>1" rules.
Please, you add the error message to the webconf rule builder ;-)

jofemodo, 2019-11-30:
OK! I'implemented N=>1 mapping. The 1=>N mapping is quite different and more complex to implement with the current code.

jofemodo, 2019-11-30:
Just update!

closed 2019-12-03 by jofemodo