zynthian/zynthian-ui #136: Setting Meter to CPU Usage stops MIDI connections from working

Baggypants, 2019-06-15:
**Describe the bug**
If you choose "CPU Usage" as the meter in the web interface, you can no longer connect synth engines to MIDI channels in zynthian-ui

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Web Config page > Interface > UI Options 
2. Click on Meter
3. Select 'CPU Usage from combo box. 
4. Click Save, the zynthian UI should reboot
5. In zynthian-ui select New Synth Layer > Pick a Synth Engine > Choose a MIDI Channel

**Expected behavior**
The UI should progress to the next screen. 

**Actual behaviour**
The UI stalls, showing the scrolling logo, buttons become unresponsive.

**Setup:**
 - SD-image: Aruk
 - Hardware Config: Kit V2
 - Software Version
```
SOFTWARE
zyncoder: master (80862b6)
zynthian-ui: master (0da8ff0)
zynthian-sys: master (3572e7e)
zynthian-data: master (676c32a)
zynthian-webconf: master (950c16b) 
```

**Additional context**
from `journalctl -u zynthian`
```
Jun 14 20:11:37 zynthian startx[518]: ERROR:zynautoconnect.zynthian_autoconnect:ZynAutoConnect ERROR: Port 'jackpeak:input_a' not available

```

jofemodo, 2019-06-15:
Update and try ;-)

closed 2019-06-15 by jofemodo

Baggypants, 2019-06-15:
Confirmed working.