zynthian/zynthian-ui #140: Enable buster

mheidt, 2019-07-02:
INFO:root:ZYNTHIAN-UI CONFIG ...
INFO:root:Wiring Layout MCP23017_EXTRA
DEBUG:root:ZYNCODER A: [102, 105, 110, 113]
DEBUG:root:ZYNCODER B: [101, 104, 109, 112]
DEBUG:root:SWITCHES layout: [100, 103, 108, 111, 106, 107, 114, 115]
DEBUG:root:DISABLED_MIDI_IN = ['']
DEBUG:root:ENABLED_MIDI_OUT = ['ttymidi:MIDI_out']
DEBUG:root:ENABLED_MIDI_FB = ['']
INFO:root:STARTING ZYNTHIAN-UI ...
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
libjackpeak failed to start jack client: 17
./zynthian.sh: line 80: convert: command not found
./zynthian.sh: line 81: xloadimage: command not found

mheidt, 2019-07-02:
this needs to be fixed first
https://github.com/zynthian/zynthian-sys/issues/110

jofemodo, 2019-07-10:
Everything seems to be working in my "patched" SD-image, but i'd to change framebuffer device file to "/dev/fb0" instead of "/dev/fb1". I'm trying to understand what changed in buster and revert the situation to use "/dev/fb1" again .... we don't want to see the boot log in the TFT display ;-)