zynthian/zynthian-ui #142: Pure Data won't make any sound until UI is displayed via a X-Window on a PC

MrDham, 2019-07-22:
**Describe the bug**
Pure Date presets don't make any sound if Zynthian is not connected to a PC and if UI is not displayed on this PC (by using "ssh -Y root@zynthian.local")

**To Reproduce**
Steps to reproduce the behavior:
Case 1:
Connect Zynthian to an audio amplifier
Create a new layer with PD > generative/generative_relaxing

Result: No sound can be heard

Case 2
Connect Zynthian to audio amplifier
Connect a PC via ethernet and log to a X-Window using  "ssh -Y root@zynthian.local"
Create a new layer with PD > generative/generative_relaxing
Result : The UI is dispayed on the PC and some automatically generated sound can be heard

**Expected behavior**
We should hear an automatically generated sound in both case 1 and 2. 
But we have some sound in case 2 only. 

**Screenshots**
N/A

**Setup:**
 - SD-image
Aruk RC3
 - Hardware Config
Raspberry Pi 3 Model B Plus Rev 1.3
Soundcard: HifiBerry DAC+
Display: PiScreen 3.5 (v2)
Wiring: MCP23017_EXTRA
 - Software Version
zyncoder: master (80862b6)
zynthian-ui: master (3688c73)
zynthian-sys: master (26f86d2)
zynthian-data: master (676c32a)
zynthian-webconf: master (0094c99)

**Additional context**
PC conf for X-Window: Xming + Putty on a windows PC (for case 2)

jofemodo, 2019-07-22:
It should be solved now. Update and test, please.

closed 2019-07-22 by jofemodo

MrDham, 2019-07-22:
I updated and tested. It works well now. Thanks.