zynthian/zynthian-ui #145: Sym-link usb directories

mheidt, 2019-08-11:
When a zynthian folder is found in the usb mount, link everything in that folder as symbolic link.
Only if the directory is not found of course.

Caveat: Symlinking and subdirectories:
/zynthian/zynthian-mydata/soundfonts/sf2/xyz/abc
should set the symlink at xyz

riban-bw, 2019-08-12:
Maybe it would be better for Zynthian to look for data in a similar directory structure on the USB mount as well as local directories. This way you don't need to worry about symlinks and supplement rather than replace internal files

mheidt, 2019-08-12:
Didn't I say that? I was only covering the case of complete snapshot folders with subfolders or preset folders with unknown subfolder structure

mheidt, 2019-09-04:
This is my suggestion:

root=/media/usb0/zynthian
root_length=${#root}

walk_directory(){
  for file in $1/*
  do
    if [[ -d $file ]]; then
      echo ${file:root_length}
      if [ -d "/zynthian/${file:root_length}" ]
      then
        walk_directory $file
      else
        echo "ln -s $file /zynthian${file:root_length}"
        ln -s "$file" "/zynthian${file:root_length}"
      fi
    else
      if [ -f "/zynthian/${file:root_length}" ]
      then
          echo "The File /zynthian${file:root_length} exists and will be ignored"
      else
         echo "ln -s $file /zynthian${file:root_length}"        
         ln -s "$file" "/zynthian${file:root_length}"
      fi
    fi
  done
}

walk_directory $root

@jofemodo  
Please test and integrate it somewhere in zynthian.sh or  as subscript of it.

jofemodo, 2019-09-18:
I've implemented this in a very simple & reliable way. 
Now Zynthian UI  lists & load  presets & soundfonts from an USB drive, combining the list with the internal sources. I would like to extend this to snapshots, but it slighty more complex and i prefer to do it in a separated commit. Also, perhaps we should consider to implement self-contained snapshots (including everything used in the snapshot).

mheidt, 2019-10-13:
I don't think that self contained snapshots is a good idea... size, duplication...

Maybe external snapshots. Those snapshots resides on the USB and they must not reference to internal my-data  resources, only our factory sounds and external ones on the usb.

If a snapshots contains my-data or usb material, we could offer a externalise feature, so that everything is copied to the USB stick.

closed 2019-11-15 by jofemodo