zynthian/zynthian-ui #147: Audio routing doesnt get saved correctly in snapshots

score100, 2019-09-03:
This happens when you have minimum 2 layers with each one or more effects on it and save a snapshot. More likely (?) to happen when you have more layers with fx. When being loaded as a snapshot they will play no sound. If looked at layer options from the layers there is no cross on either system or the attached effect.

**To Reproduce**
Steps to reproduce the behavior:
1. Create 2 layers.
2. Add an effect (I use zynverb) to each of them, save snapshot
3. Power off, and power on. Load snapshot.
4. Play some notes and hear no audio. Looking at the audio routing you can see it ain't routed to nowhere.

**Expected behavior**
You'd expect the zynthian to save the way audio is routed as well.

**Setup:**
Using Aruk RC-3. Last update 8 days earlier.

**Additional context**
This one is kinda tricky as it happened multiple times to me while preparing for a concert. Sometimes all layers get disconnected. Sometimes only one.

Arxi, 2019-09-15:
Can confirm this issue, and I'm unable to route the audio manually either - it seems like the audio effects in a layer get disconnected, or maybe it's better to say unchained. @score100 have you been able to find a reliable way to "re-set" the layer audio routing to intended state after snapshot is loaded, at least manually??

score100, 2019-09-16:
@Arxi  so far I could do so manually by using the layer options. 
Generally this one is kind of difficult to reproduce. I had it also sometimes not happening at all. Then again a few times in a different way, where a new layer would be automatically connected to an exisiting effect.

jofemodo, 2019-09-16:
I've commited some fixes tat should solve the probles. Please, update software and test ;-)

Arxi, 2019-09-17:
@jofemodo from what I've tested so far, seems like the audio is now routed correctly after the snapshot is loaded. Great job, and thank you very much!

Arxi, 2019-09-24:
@jofemodo  I managed to come across this bug again - see the attached .zss file. When I load this snapshot, sometimes the audio is not routed anywhere (no sound); sometimes it is routed correctly, but when I leave the Control view and go to layer list, the effects are suddenly bypassed and the synth itself is "reset" to its default values.

It might be connected to the way I created this snapshot: I created 6 layers one by one, each on its own midi channel. Then I deleted the first 5 layers, leaving only the layer 6. Then I tried to move the layer 6 to midi channel 1 - I wasn't able to get any sound from it afterwards (sending notes on channel 1 from sequencer of course), so I moved it back to midi channel 6.

I have the latest updates (updated just now).

[007-1.2-Trip.zss.zip](https://github.com/zynthian/zynthian-ui/files/3648430/007-1.2-Trip.zss.zip)

Arxi, 2019-09-24:
Hmm. Seems like the synth reset thing might be a separate issue. When I create a new clean MDA JX10/Soft E.Piano layer, change sam parameters and then go back, the parameters are reset to their defaults.

jofemodo, 2019-10-02:
The problem when moving a layer to a different MIDI channel should be fixed now ...

Arxi, 2019-10-11:
I updated to the latest version, but now it's kinda hard to test - while loading my older snapshots, most of them say e.g. `JV/MDA JX10 > NoBank` and look like no preset is even selected. When I select a preset, the audio rouitng is incorrect, too. Or should I recreate the snaphots again with latest updates?

Arxi, 2019-10-11:
So I created a new snapshot, very minimal, just to test the audio routing. Saved, loaded. Audio isn't routed anywhere. Here's the snapshot:
[013-1.3-Vianocka.zss.zip](https://github.com/zynthian/zynthian-ui/files/3719267/013-1.3-Vianocka.zss.zip)

jofemodo, 2019-10-14:
Hi @Arxi !
Could you try this snapshot?

[014-Jofenocka.zip](https://github.com/zynthian/zynthian-ui/files/3725236/014-Jofenocka.zip)

Regards,

jofemodo, 2019-10-14:
Regarding your "Vianocka" snapshot, there are 2 problems:
+ We have reorganized Dexed (and other LV2 plugins) presets in banks, so the you have to re-select the bank & preset and save snapshot again.
+ The layer is in ch#5 and "single channel mode" is disable (FYI, MIDI profile options are now saved/restored with snapshots), so perhaps this is the "routing problem".

Regards,

jofemodo, 2019-10-14:
Hi @Arxi !
I think i've fixed an old problems. Finally i understanded why trying to reuse engine instances between snapshots is failing. I've disabled this "optimization" until i can write a better algorithm for it.
I hope this solves a lot of snapshot-related problems.

The bad part is that snapshots saved after loading several snapshots probably are wrong and should be rebuilt from scratch because routing is broken on them. Broken snapshots can't be recovered easily and i would prefer to not add extra code for recovering these snapshots. Sorry.

commit 880b5905 (2019-09-16): Fix #147: Fix autoconnect and audio-routing methods for correctly work when loading snapshot with several layerss.
commit a56f74b5 (2019-10-14): Fix #147: Audio routing doesn't work correctly when loading a snapshot over another one.
Trying to reuse engine instances is problematic because jack names doesn't fit the saved ones and audio routing fails.
closed 2019-11-15 by jofemodo