zynthian/zynthian-ui #148: Record Status does not indicate MIDI & or Audio Record.

wyleu, 2019-09-03:
**Is your feature request related to a problem? Please describe.**
If Audio Recording (Kbd R) is enabled , followed by MIDI Record (Kbd M), a record status is displayed which behaves correctly. (If you turn off Audio Recording (Kbd Shift R) the record status is still displayed cos MIDI recording is still enabled). 
It would be useful if the states could be differentiated possibly in the design of the RED Circle

**Describe the solution you'd like**
Audio Record represented by a large Red Circle with a missing black circle on the south sector of the outer circle. 
MIDI record fills the lower defined black circle so when both are recording, a full red circle are displayed.
I would change the default for audio record to record a MIDI track as well and the Key Binding becomes A/Shift A for audio only recording.

**Describe alternatives you've considered**
Keep a good mental image of what you are recording when.

riban-bw, 2019-11-14:
I did consider how to differentiate between MIDI and audio recording when I worked on this but the small icon space did not lend itself to this, especially on my very small screen. We could use different colours but that might not be intuitive. We could use different border colour (for the red dot) but similar issue with intuitiveness. A letter within the red circle may work on larger displays.
We could use the space occupied by the MIDI received data indication 'M' for MIDI record / play status but this may require using a different display method for MIDI receive. I did implement an icon for that but @jofemodo reverted to the 'M' so I guess it isn't preferred. (I don't like the 'M' - it looks out of place in a notification area dominated by icons! I would prefer some nicely crafted icons for all notifications.)

jofemodo, 2019-11-15:
I also prefer an icon for MIDI input notification, but i didn't like the icon you put, and i couldn't find a better one ... so i put the M, that is quite intuitive, although i agree with you: it's not nice and an icon should be preferred ;-)

Regarding the issue's subject, i also thought about the problem but i didn't find a good solution ... by now.

jofemodo, 2019-11-15:
Although it's not a "perfect" solution, perhaps using "blue" for MIDI record/play icons could be intuitive, the "m" is also blue. It could be a provisional solution ... what do you think? @wyleu ?

riban-bw, 2019-11-15:
But what about concurrent MIDI and audio recording? Purple???

jofemodo, 2019-11-15:
As i told you, it's not a perfect solution. Concurrent audio/MIDI recording is not a normal use-case. Zynthian is not a DAW ... yet ;-)

riban-bw, 2019-11-15:
Yep, not ideal but we need to know what to display for that condition. This is the very reason I left it as a common red dot because anything else needs a rule-set to define behaviour under each condition. I work a lot with accessibility so am always weary of using just colour to distinguish between states but the use of colour to reinforce a user's view is good 🙂.

wyleu, 2019-11-16:
The red Dot seems simple and effective and visible from some distance away. Blue dot works for MIDI and matches the Blue M ( I tend to agree with riban about the M but I'm used to it by now. It's probably one of the few words (or acronyms if you prefer)  that actually has the same meaning across all western languages. 
For MIDI & audio red dot with a blue border? probably easier to see than Purple.

wyleu, 2019-11-16:
Fully understand the rewrite but I would suggest even in it's simple form recording has proved a boon.

closed 2020-05-09 by wyleu