zynthian/zynthian-ui #153: Zynthian Audio Play needs Volume Window

Mickinator, 2019-10-02:
The Zynthian Admin/Audio Recorder needs a window called "Volume" so the user can set a dedicated volume while an audio file is playing,
Also the user can adress this field to a MIDI-Controller Number to control the volume with an external controller.

jofemodo, 2019-12-05:
Hi @Mickinator and others!
Please, update your zynthian  and test the "Playback Volume Control".

Mickinator, 2019-12-06:
> <mailto:notifications@github.com> >
> (#153)
> <mailto:zynthian-ui@noreply.github.com> >
> <mailto:norbert.schreiber@t-online.de> >, Mention <
> mention@noreply.github.com <mailto:mention@noreply.github.com> >

Hi Fernando,
great, wonderful❤️

but I cant set a midicontroller
do I something wrong?

von Norbert Schreiber gesendet

jofemodo, 2019-12-06:
I Norbert! The audio playback volume can't be assigned to MIDI currently.
I will see if i can so something about ... ;-)

Mickinator, 2019-12-06:
👍

von Norbert Schreiber gesendet

Datum: 06.12.2019, 23:51 Uhr

closed 2019-12-20 by jofemodo