zynthian/zynthian-ui #154: Performance Audio-or Miditracks

Mickinator, 2019-10-04:
To have a propper gig-performance it should be good to have

- title-storing into snapshot
- volume Window for audio/Midi-track
- transport-buttons for
   - play
   - stop
   - rewind 
(all midi-CC-controllable).

mheidt, 2019-10-04:
Have you tried to give it a title I the webconf?

Mickinator, 2019-10-04:
I had a conversation about the transport knobs with jofemodo

jofemodo, 2019-10-06:
Have you tried to give it a title I the webconf? ;-)

jofemodo, 2019-10-07:
I added CUIAs for play/stop MIDI & Audio tracks.
Also implemented MIDI System Control messages for MIDI track play/stop.

Mickinator, 2019-10-11:
> <mailto:notifications@github.com> >
> <mailto:zynthian-ui@noreply.github.com> >
> <mailto:norbert.schreiber@t-online.de> >, Author <
> author@noreply.github.com <mailto:author@noreply.github.com> >

Hi Fernando,

what have I to do, to work with this new features?
I can send Note or controller or prgrch- Commands. But what numbers?
Where can I read it?

Thanx for helping!
Micki

Gesendet mit der Telekom Mail App
<http://www.t-online.de/service/redir/emailmobilapp_ios_smartphone_footerlink.htm>

jofemodo, 2019-10-11:
In the wiki's documentation, my friend ;-)

http://wiki.zynthian.org/index.php/Zynthian_UI_Users_Guide#Controlling_the_UI_with_MIDI_messages

Mickinator, 2019-10-11:
Wow, that's great.

I only knewed the old one. This evening I have a lot to do!
Thank you so much, my friend 😘🎶🌈

Gesendet mit der Telekom Mail App
<http://www.t-online.de/service/redir/emailmobilapp_ios_smartphone_footerlink.htm>

Datum: 11.10.2019, 13:52 Uhr

http://wiki.zynthian.org/index.php/Zynthian_UI_Users_Guide#Controlling_the_UI_with_MIDI_messages
<http://wiki.zynthian.org/index.php/Zynthian_UI_Users_Guide#Controlling_the_UI_with_MIDI_messages>

Mickinator, 2019-10-15:
I´ve tried it, but I have no success.
Where are the files I like to play? I see nothing in the Audio Window or in the Webconf Screen.
Where can I choose the wanted song?

Mickinator, 2019-10-15:
I only see self recorded captures
but no files from the usb-stick

jofemodo, 2019-10-18:
Volume window is not implemented yet, and it will take some time as it's not a trivial task. Please, be patient.
Regarding the USB-stick, files have to be ".wav" files in the root dir, Files in subdirs will not be shown.

closed 2019-11-26 by jofemodo