zynthian/zynthian-ui #157: Audio In needs Control-window

Mickinator, 2019-10-07:
Hi Zynthianers,

since I use a behringer uca222 with raspt4 I need an Audio-In-Control-window with

- volume
- expression
- pan
- effects
- small eq

Thanx for helping
Micki

Mickinator, 2019-10-09:
In addition: I think all engines should receive the controller #7, 11, 64, pan, by default regardless there is a window or not.

Mickinator, 2019-11-08:
For a first step:
Is it possible to route the Audio-In to Audio-Out and merge with the Zynthian-Sound?

jofemodo, 2019-11-08:
Of course. Create a new "Effect Layer" and select "Gain 2x2" as engine.

Mickinator, 2019-11-08:
Did I say that I love you?😂😂😂😂😂

Mickinator, 2019-11-08:
Is it possible that the Midi-learn-function not works with Gain 2x2?
I cannot adress Ch#15 CC#7 ...

jofemodo, 2019-11-08:
Master MIDI channel can't be learned, sorry. You have to use a different channel

Mickinator, 2019-11-08:
> <mailto:notifications@github.com> >
> <mailto:zynthian-ui@noreply.github.com> >
> <mailto:norbert.schreiber@t-online.de> >, State change <
> state_change@noreply.github.com <mailto:state_change@noreply.github.com> >

Yes, of coarse.
Masterchannel is Ch 16 (1 - 16)
This is Instrument Ch 15

Gesendet mit der Telekom Mail App
<http://www.t-online.de/service/redir/emailmobilapp_ios_smartphone_footerlink.htm>

jofemodo, 2019-11-08:
Ups sorry! Seems to be a bug .... let me solve it ;-)

jofemodo, 2019-11-08:
Ups ... no! It works like a charm for me.
Please, could you describe the problem exactly:
- SD image version
- how did you try to learn
- etc.

Mickinator, 2019-11-08:
The Prob is soled (nearly).

I do not use the right display driver. I have a
MAKERFACTORY VMP400 Touchscreen-Modul 8.9 cm (3.5 Zoll) 320 x 480 Pixel , not a 
WaveShare 4 GPIO-only. So the touchscreen features don´t work well. Is the WaveShare 4C GPIO-only better? What is the difference?

But I could solve my prob using the mouse.
Now I can control my playbacks from the tablet with the same faders as the kb-sounds.
Won der ful !!!
[image]

Mickinator, 2019-11-09:
Hi Fernando,
I believe Audio In is only working with 1 channel. Is this possible?
What can I do to hear stereo?

jofemodo, 2019-12-05:
Currently, each audio input channel is routed to the same output channel.
You can force to mix both channels by including in the "FX chain" a mono-plugin. For instance, the "Gain 1x1" (probably you have to enable the plugin from the webconf).

jofemodo, 2019-12-05:
Regarding the subject of the issue, i'm implementing a "Mixer" panel that allow to modify the ALSA mixer levels from the UI. Currently it's in the development branch "mixer".

Also this issue is being adressed by #175 ...

closed 2019-12-20 by jofemodo