zynthian/zynthian-ui #159: Pianoteq and C.Bechstein

mheidt, 2019-10-13:
I bought the BechsteinDG:A licence and found a nice bug.

The issue is, that the presets of the old C. Bechstein and the new C. Bechstein DG are in the same tag.

"C. Bechstein": [["C. Bechstein DG Prelude", null, "DG Prelude", null], 
....
 ["C. Bechstein DG Broken", null, "DG Broken", null], ["C. Bechstein", null, "<default>", null], ["C. Bechstein Close Mic", null, "Close Mic", null], ...

The problem that we have is, that the C. Bechstein only appears once and in the demo section.
If I select a C.Bechstein DG instrument, I can play all keys and if I chose a NON DG bank, I have the demo behaviour. Like expected.

Personally I think, this is a Pianoteq issue and they should manage their preset list and divide them. But I think, that they won't.

I am not quite sure, how we should tackle it.

Changing bank_list_v6_4 = [
		('BechsteinDG', 0, 'BechsteinDG', '_', 'BechsteinDG:A')
	]

to

bank_list_v6_4 = [
		('C. Bechstein', 0, 'BechsteinDG', '_', 'BechsteinDG:A', 'C. Bechstein DG')
	]

where the last value is a filter of the instruments. And we should use that last parameter instead of the first as bank name?

Do you agree?

mheidt, 2019-10-13:
Or what about bank[2]?
This is only used for logging right now?

mheidt, 2019-10-13:
Ok, the solution is rather in
logging.info("Caching Internal Presets ...")
		#Get internal presets from Pianoteq ...
		try:
			pianoteq=subprocess.Popen([PIANOTEQ_BINARY, "--list-presets"],stdout=subprocess.PIPE)
			for line in pianoteq.stdout:
				l=line.rstrip().decode("utf-8")
				logging.debug("PRESET => {}".format(l))

mheidt, 2019-10-13:
If I change the bank definition to
bank_list_v6_4 = [
		('C. Bechstein DG', 0, 'C. Bechstein DG', '_', 'BechsteinDG:A')
	]

This bank has the correct instrument selection.
But C. Bechstein has all :)

Solution is close...

mheidt, 2019-10-13:
PushRequest....

mheidt, 2019-10-13:
Maybe the banklist needs to be sorted by length first, so that the longer bank names are considered first?

jofemodo, 2019-10-14:
Merged! It should solve the problem by now ... although as you say, Pianoteq should add a column with the "Instrument" when listing presets ... I will try to convince them ;-)

closed 2019-10-14 by jofemodo