zynthian/zynthian-ui #162: OSC / Load Snapshot / Create Snapshot

mheidt, 2019-10-18:
I am currently working on a screenless raspbi4 setup.
I don't know how to load a Pianoteq layer into the system.
It would be nice, if I could create one over webconf and load it.

Once the API exists, I could integrate it into webconf.

jofemodo, 2019-10-18:
Completely OK.
Currently i'm working on the presets API, so i could continue with this ;-)

mheidt, 2020-01-24:
I think, here a REST API in the zynthian-ui process would be helpful.
Like I needed for the audio-zcontroller pages.

mheidt, 2020-02-29:
Right now I help my self using UIActions. 
The only issue is that a BOLD Snapshot press toggles between Load and Save which is quite risky on a screenless device.
I would see two options besides the API call which this task is for.
A separate UI Action which shows the LoadSnapshot and doesn't toggle.
This might the easiest, but it's not really good.
Better would be a LoadSnapshotXYZ Action.
In the snapshot area you can define if a snapshot is loaded with CustomActionX(Normal,Bold,Long)
In the wiring it is only displayed or reset to a different acion type.
Then you would be able to build a screenless zynthian-module with a MCP23018 and 4 preset buttons.

mheidt, 2020-03-04:
@jofemodo, I would have a urgent request.
Is it possible to define a
Long LS push here
https://wiki.zynthian.org/index.php/File:Zynthian_ui_workflow_snapshots.svg

This is always going to Load Snapshot and preferably selecting the first real snapshot (not default or the like). And not the one that was loaded the last.

jofemodo, 2020-03-04:
Sorry @mheidt , i don't understand your request.
Could you explain it again? ;-)

mheidt, 2020-03-04:
Imagine a screen less environment. I have 3 UI Actions defined.
One is LS. But as you know, those buttons don't necessarily work the best physically.
Sometimes you press without any result. With screen, you would press again.
But in this case a second press would go to SAVE screenshot...deadly on stage...
I want a 100% secure action that guarantees the LOAD snapshot screen no matter how often I trigger it.
My second button is a "Selection down", it would be of advantage if I always start on the first selection when this new Action (Long LS) is pressed.

Besides, i have the impression that when you were on SAVE screen and loaded a snapshot and press normal LS, you start at Save again.
But I am not quite sure. I am ending at Save far too often if I consider that I only pressed LS once.
I defined Normal and Bold of this action to be Normal LS, to ensure that I didn't press it too long, btw.

mheidt, 2020-03-04:
ah,@jofemodo just in case you don't get emails after i commented

jofemodo, 2020-03-04:
Try now ... ;-)

mheidt, 2020-03-04:
Yes, perfect. Forget what I said about "start on the first selection".
It's better the way it always was..selecting the loaded snapshot...

Thanks a lot for the quick answer..rehearsal is tomorrow :)

Nevertheless this issue is about direct loading snapshots with actions. But this is low prio now.

mheidt, 2020-03-05:
To summarize what is needed, especially in regards to 
https://github.com/zynthian/zynthian-webconf/issues/123

- Create empty snapshot (should be easy or exist already)
- CustomUiAction to load a snapshot (that what I mentioned 5d earlier)
- When the REST interface exists
-- Load snapshot into zynthian-ui

mheidt, 2020-03-06:
@jofemodo 
If I understand correctly zynapi is limited to engines right now.
If I want to use 
zynthian_gui_layer.save_snapshot(self, fpath) 
in webconf.

How should we do that?

mheidt, 2020-03-07:
@jofemodo 
If the new CustomAction ProgamChange hat the option to set the midi channel (esp. to master channel), we don't need what I suggested rgd loading snapshots with CustomAction.
Is that technically possible or are we restricted on current midi channel here?

jofemodo, 2020-03-08:
OK! It's implemented in 
https://github.com/zynthian/zynthian-ui/commit/5ccaeec7d31647476e3b01c15c0b7beaf3780fdd
and
https://github.com/zynthian/zynthian-webconf/commit/a7f569fe7d8a0a373103b3c1fb3c200f4b17517f

mheidt, 2020-03-09:
My Testcase:
Two snapshots at 000/001 and 000/002
MasterChannel 16
MasterChangeType Custom with MSB (default, no other values)
Then I defined a Midi Program Change to 16/1 for CustomSwitch2

I would except one of the two snapshots to be loaded...still curious which one though :)

But nothing happened. 
Debug Log:
Mar 09 00:32:21 zynthian startx[1030]: INFO:zynthian_gui: Short Switch 5
Mar 09 00:32:21 zynthian startx[1030]: WARNING:zynthian_gui: 'S'

shouldn't the label of the action be MIDI instead of Midi?...hmmm..Midi filter rules as well...
Did you cover a Bold switch? Sometimes one might press a little too long.

mheidt, 2020-03-09:
@jofemodo 
I made some tests with NoteOn as well.
This doesn't work either.
Simple zynaddsubfx layer at channel 1
Action 2 got a NoteOnAction

Mar 09 22:40:51 zynthian startx[1377]: INFO:zynthian_gui: Looooooooong Switch 5
Mar 09 22:40:51 zynthian startx[1377]: WARNING:zynthian_gui: 'L'
Mar 09 22:40:55 zynthian startx[1377]: INFO:zynthian_gui: Bold Switch 5
Mar 09 22:40:55 zynthian startx[1377]: WARNING:zynthian_gui: 'B'

jofemodo, 2020-03-10:
Try now ;-)

mheidt, 2020-03-13:
now that you solved it, those are the ones left

- Create empty snapshot (change existing function as a static class method)
- (When the REST interface exists) Load snapshot into zynthian-ui

closed 2021-11-30 by mheidt