zynthian/zynthian-ui #163: Enable CUIA for MCP23008

mheidt, 2019-10-20:
https://discourse.zynthian.org/t/minimal-ttymidi-in-wiring/3340/9

Define a new wiring forMCP23008, that offers the CUIA action options in webconf.

In fact, code for MCP23017 should be extended to support MCP23008 too, and former MCP23008 code should be removed.

jofemodo, 2019-10-30:
Hi @mheidt !

I've been thinking about it and it's not possible to use the MCP23017 for the 2in1 module because this module have not connected the INT-A pin, so pins should be polled, as it's already done by the current working code for the MCP23008.

In fact, you can already use it by simply configuring the wiring from the webconf tool.

I just added a few lines to the wiki, at the end of this section:

http://wiki.zynthian.org/index.php/Adding_Extra_Switches

closed 2019-10-30 by jofemodo

mheidt, 2019-10-30:
And I have to use the 3.3V?

jofemodo, 2019-10-30:
Yo need both, 5v and 3.3V. This module is not very well designed ;-)

Follow the Kit v1 instructions:

http://wiki.zynthian.org/index.php/Building_a_Zynthian_Box_using_an_official_Kit#Connection_Tables

mheidt, 2019-10-30:
> To GND, my friend. It's documented in the wiki:
> http://wiki.zynthian.org/index.php/Adding_Extra_Switches
> FYI, all switch-pins in zynthian are "pulled up":
> https://github.com/zynthian/zyncoder/blob/8b7bfbdaff11e984e1c0e06d480de03a7f1eef3f/zyncoder.c#L338
> So, you have to connect the other pin to GND.
> Regards,

Lol
Sorry, I am software developer :) Wednesday,