zynthian/zynthian-ui #164: UI Freeze after RemoveAll Snapshots

mheidt, 2019-10-20:
I loaded a snapshot with 5 layers successfully.
But when I RemoveAll layers, the UI freezes.

mheidt, 2019-10-20:
[error.log](https://github.com/zynthian/zynthian-ui/files/3748379/error.log)

mheidt, 2019-10-20:
The last 3 lines are normal shutdown behaviour:

xinit: connection to X server lost
waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.
xinit: unexpected signal 2

jofemodo, 2019-10-21:
Hi @mheidt !

Could you send a snapshot that reproduce the problem?

Thanks!

mheidt, 2019-10-21:
Only tonight ..doesn't the error.log print the snapshot as well?

mheidt, 2019-10-21:
I find it in the dropbox

mheidt, 2019-10-21:
But I cannot reproduce it anytime anymore.
Try, loading it, BackBackBack.... to LayerList, RemoveAll, Load it Again immediately. (when I pressed RemoveALL twice, the next time worked). BackBackBack... to LayerList, RemoveAll, Freeze

closed 2019-10-30 by jofemodo