zynthian/zynthian-ui #166: Master Program Change / MIDI_BANK_DIMENSION

mheidt, 2019-10-25:
As I understand correctly, the folders of the snapshots are organizing the banks.
The master keyboard can send CC 00 or 32 and depending on the MasterBankChangeMode (LSB,MSB), 00 or 32 is considered.

But the real world is a little more complex.
The D50 is organized in 2x8x8 and not sending any CC 00/32.
The Kawai MP 7 SE is organized in 8x8x4. The first 4 of the first dimension is CC32=0 and the second is CC32=1.

Really cool would it be, if we could map this.
I think, a 100% solution where the directory structure of snapshots is exactly like the button structure of the keyboard, is too much to ask for.

But we could add more directory levels.
We just need ony more parameter MIDI_BANK_DIMENSION or the like.
The D50 would need a value of "2x8x8" and the Kawai "4x8x4". Default would be "128".
In both cases we would allow a snapshot directory depth of 4.
The first for the CC bank and 3 for the program changes.
Snapshots can only be stored in the deepest folder.
We would create the directory structure and the webconf can rename them as usual. But they cannot be deleted .
Snapshots can only be stored in the deepest directory.
The user is storing them like now...into a program change of 0..127
The webconf calculates, into which subdirectory it has to be stored.
Should we allow a change of the dimension? In that case we have to move the snapshots.

When the UI can handle subdirectories, this task can be moved into webconf.

jofemodo, 2019-10-28:
Hi @mheidt !
As you probably remember, i disabled the snapshot banks by default, because it was too confusing for most of people, and very little people use this feature. For most of people, having all the snapshots in the same directory is perfectly OK, given that they can rename the snapshots from the webconf tool.
From my POW, adding more directory levels is not the way to go. Browsing multiple directory levels from the Zynthian-UI is not comfortable and i would like to simplify the UI, not making it more complex. Same for the webconf.
In the other hand, i would like to allow a good integration with keyboards and controllers, and this include allowing to map their bank/program structures. Also, i would like that you could use the same Zynthian with several keyboards and have different mappings for every keyboard. 

OK. We are far from this currently, but i would like to advance in that direction ... so .... let's think a little bit about it ;-)

As starting point, I would suggest to store separately any "device dependent" mapping structure.

Finally ... i'm curious  about how are you using ZS3 subsnapshots in your workflow ...

Regards,

mheidt, 2019-10-28:
In this case, we need to disable additional banks in the webconf!
I was using it...
And I don't know, how you are going to handle different banks, that are sent by the keyboard.
I agree, that we can skip this feature though.
But we need bank handling.

jofemodo, 2019-10-28:
I don't think so. Webconf's tool maps snapshots to standard MIDI bank & program numbers. This is device-independent. The problem is how some devices map these standard MIDI banks & programs to their buttons & pads.

mheidt, 2019-10-28:
ok, i agree. The feature is too complicated

closed 2019-10-28 by mheidt