zynthian/zynthian-ui #168: MIDI-MAP ProgramChange

mheidt, 2019-10-26:
My goal is to Map all Program Changes from all channels to master channel 15

I found several bugs:
1. ERROR parsing MIDI filter rule: Invalid MAP rule format (MAP PC => CH#15 PC): MAP rule channel lists can't be matched

When I try to be specific:
MAP CH#0 PC => CH#15 PC
MAP CH#1 PC => CH#15 PC

The program change is not routed, so that a snapshot program change works.
When I send to channel 16 (aka15) directly, the snapshot is changed.

mheidt, 2019-10-26:
I correctly sent message through channel 16

INFO:root:MIDI_UI MESSAGE: 0xbf0000
INFO:root:MIDI_UI MESSAGE DETAILS: 15, 11
INFO:root:MASTER MIDI MESSAGE: 0xbf0000
INFO:root:MIDI_UI MESSAGE: 0xbf2000
INFO:root:MIDI_UI MESSAGE DETAILS: 15, 11
INFO:root:MASTER MIDI MESSAGE: 0xbf2000
INFO:root:MIDI_UI MESSAGE: 0xcf0568
INFO:root:MIDI_UI MESSAGE DETAILS: 15, 12
INFO:root:MASTER MIDI MESSAGE: 0xcf0568

the one through channel 1
INFO:root:MIDI_UI MESSAGE: 0xb00000
INFO:root:MIDI_UI MESSAGE DETAILS: 0, 11
INFO:root:MIDI_UI MESSAGE: 0xb02000
INFO:root:MIDI_UI MESSAGE DETAILS: 0, 11
INFO:root:MIDI_UI MESSAGE: 0xc00668
INFO:root:MIDI_UI MESSAGE DETAILS: 0, 12
INFO:root:MIDI PROGRAM CHANGE: CH0 => 6

It is not recognized as MASTER MIDI MESSAGE. Although earlier:

MAP CH#0 PC#0 => CH#15 PC#0
DEBUG:root:MAP CH#0 PC#1 => CH#15 PC#1
DEBUG:root:MAP CH#0 PC#2 => CH#15 PC#2
DEBUG:root:MAP CH#0 PC#3 => CH#15 PC#3
DEBUG:root:MAP CH#0 PC#4 => CH#15 PC#4
DEBUG:root:MAP CH#0 PC#5 => CH#15 PC#5
DEBUG:root:MAP CH#0 PC#6 => CH#15 PC#6

Mickinator, 2019-10-26:
Hi Markus,
I think to map all PC-Commands to the masterchannel cannot be the goal. It may be a nice to have.
The main thing is, that we can adress a specific bank via cc#0 and cc#32 and then we can send a PC-command to select a snapshot. 
In my case it would be possible to change entries of a snapshot if I use other channels. I think, that would be a great option.

mheidt, 2019-10-26:
@Mickinator 
It's optional. I need to map all Program Changes to Channel 15.

jofemodo, 2019-10-28:
Currently, for mapping "all program changes to channel 15", you have to use a rule for every channel. Sorry. The rule parse is quite limited right now.

jofemodo, 2019-10-28:
Regarding the Master Channel routing itself, currently Master Channel messages are captured before applying rules, so you can not route messages from other channel to Master Channel with the current behaviour. I can't remember why i implemented in such a way and it's not documented, so i'm thinking about the implications of moving downstream the MC capture ...  

Regards,

jofemodo, 2019-10-28:
OK! I remember. Master Channel events must be captured before "Active Channel" conversion, and this is also done before applying rules.
Now, i've to remember why i implemented the "Active Channel" conversion before applying rules ... jeje!

jofemodo, 2019-10-28:
OK! I remember. Moving "Active Channel, (aka Single Channel Mode)" conversion downstream will break MIDI learning and it's not easy to solve, as currently a part of MIDI-learning is implemented using rules. For solving that i should have 2 separated rule-pipes, one for MIDI-learning and another one for "user/webconf defined rules".
Sorry, but it's not a 2-minutes change ...

mheidt, 2019-10-28:
This is the usecase, I need it for.
Every snapshot contains serveral layers aka midi channels.
One snapshot per song is ok. But when you are on stage you want to load the next song by clicking on a program change trigger, no matter on what channel you ended the song with.
And you don't want to set your keyboard to master channel 16 first.

Not to speak in a single channel mode environment...altough in this case, the program change event might get routed to channel 16?

jofemodo, 2019-10-28:
Many keyboards allows to assign the pads/buttons to any combination of bank/program change in the channel you want, no matter what channel you have selected in the keyboard, so no problem with these keyboards. Most of my keyboards work in this way. In fact, all of them ;-)

But i agree that we should solve the problem for ALL keyboards ... but without making things more complex for people that doesn't need it!

jofemodo, 2019-10-28:
Take into account that Program Change is used by ZS3 subsnapshots, so we can not route it to channel 16 when in Single Channel Mode. We need program change to be mapped to the active channel... except Master Channel, that is reserved for Snapshot loading.

I'm thinking about all this ... it's not easy to keep working all the interesting use cases ;-)

mheidt, 2019-10-28:
Yes, new DAW keyboards don't have the issue. But if you have stage pianos, usually you don't have that luxury.
But I agree, that Single Channel Mode and this feature is mutual exclusive.

I think an alternative feature could solve the issue.
If you have a snapshot selected, you could choose one of the 4 MCP Extra actions.
And if you press the button, the snapshot is loaded.

I close this. Once the MCP23008 is capable of handling those CUIA actions, I will remind you again

Mickinator, 2019-10-29:
> <mailto:zynthian-ui@noreply.github.com> >
> <mailto:norbert.schreiber@t-online.de> >, Mention <
> mention@noreply.github.com <mailto:mention@noreply.github.com> >

Most Stage Pianos habe a feature to send
LSB/MSB/PC.

I Do need exactly that. Please use the LSB/MSB to specify the folder, you 
are working with.
And than use the PC to specify the snapshot.

I think, so are the Midi-rules. We have to use them!

Regards
Micki

Gesendet mit der Telekom Mail App
<http://www.t-online.de/service/redir/emailmobilapp_ios_smartphone_footerlink.htm>

jofemodo, 2019-10-29:
This is exactly what is currently implemented. You can use MSB or LSB (not both at once) to specify the bank folder and PC for the snapshot file.

Mickinator, 2019-10-29:
[You can use MSB or LSB (not both at once) to specify the bank folder ]

Why that? Please specify one controller (LSB or MSB) to do that and set the 
other to "0".
If you use the PCh logic consequently, you can use the free values for 
other purposes, e.g.
you can start an audio-file with playbacks or drumloops or gimmiks.

Btw: a lot of keyboards send the LSB/MSB/PCh tripple. If you do not use one 
value, you cannot use the PChCommand.

Zynthian has to understand a full tripple PCh-Command!

Gesendet mit der Telekom Mail App
<http://www.t-online.de/service/redir/emailmobilapp_ios_smartphone_footerlink.htm>

Datum: 29.10.2019, 10:53 Uhr

mheidt, 2019-10-29:
@micki, I was not saying that a stage piano is not sending pc, but that a stage piano doesn't have freely customizable pads.

And rgd lsb/msb, I think it works exactly like you need it. When both is send, you can define in webconf, which to take. What else should be possible?  Adding both values? Kidding...

Mickinator, 2019-10-29:
Hi Markus,
ok, I will try it tomorrow.

But to respect the midi rules is not kidding, it's better.

An Instrument may send a command in evry correct way. The receiving 
instrument has to interpret it correctly.
And see
- a pch gives you 128 possibilities
- lsb and pc give you 128 x 128 possibilities = 16384
- lsb and msb and pc give you 2 097 152 possibilities, and a great range to 
organise your data.
Thats not kidding, thats organisation.

Best regards
Micki

Gesendet mit der Telekom Mail App
<http://www.t-online.de/service/redir/emailmobilapp_ios_smartphone_footerlink.htm>

jofemodo, 2019-10-29:
Hi @Mickinator,
Please, try what is currently implemented ;-)
Thanks!

Mickinator, 2019-10-29:
Yes, sorry, I'll do that. Sorry😥👍🌈

I'm so happy to have thing, I can play with. On a professionell level.
And a lot of thanks to you and your team.💐💐💐

Gesendet mit der Telekom Mail App
<http://www.t-online.de/service/redir/emailmobilapp_ios_smartphone_footerlink.htm>

Datum: 29.10.2019, 12:28 Uhr

mheidt, 2019-10-29:
He has a point though.
Now that you have a MIDI Log so that you can see the messages, it might be better to avoid LSB and MSB, remove the switch and just implement a LSB (CC0)directory which contains MSB (CC32) directories and then the pc. 
I have to add an additional selectox in webconf.

This seems more intuitive to me.

Mickinator, 2019-10-29:
After Update (2x) and reboot:
 
WebConf is good, if it works.,
 
Now it shows nothing and I can see nothing.
 
But I need a guide:
What is my start-window? load snapshot?
My keyboards send two CC# (0 and 32). What do I do with the unused? set to 
0?
 
I like to call the snapshot "Modern Times".
I´ve choosed
Master Channel 16.
Bank 2 MMP
Snapshot 5 Modern Times
 
see picture 4
 
best regards
Micki

 
 
 

"https://github.com/zynthian/zynthian-ui/issues/168?email_source=notifications\u0026email_token=ANJNSQL7SSOJFHEYBYV5DX3QRAV6DA5CNFSM4JFMK5QKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOECQK7CY#issuecomment-547401611", 
"url": 
"https://github.com/zynthian/zynthian-ui/issues/168?email_source=notifications\u0026email_token=ANJNSQL7SSOJFHEYBYV5DX3QRAV6DA5CNFSM4JFMK5QKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOECQK7CY#issuecomment-547401611", 
"name": "View Issue" }, "description": "View this Issue on GitHub", 
"publisher": { "@type": "Organization", "name": "GitHub", "url": 
"https://github.com" } } ]

Mickinator, 2019-10-29:
2nd try picture 4 was wrong, 41 is right
 
 
After Update (2x) and reboot:
 
WebConf is good, if it works.,
 
Now it shows nothing and I can see nothing.
 
But I need a guide:
What is my start-window? load snapshot?
My keyboards send two CC# (0 and 32). What do I do with the unused? set to 
0?
 
I like to call the snapshot "Modern Times".
I´ve choosed
Master Channel 16.
Bank 2 MMP
Snapshot 5 Modern Times
 
see picture 4
 
best regards
Micki

 
 
 

"https://github.com/zynthian/zynthian-ui/issues/168?email_source=notifications\u0026email_token=ANJNSQL7SSOJFHEYBYV5DX3QRAV6DA5CNFSM4JFMK5QKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOECQK7CY#issuecomment-547401611", 
"https://github.com/zynthian/zynthian-ui/issues/168?email_source=notifications\u0026email_token=ANJNSQL7SSOJFHEYBYV5DX3QRAV6DA5CNFSM4JFMK5QKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOECQK7CY#issuecomment-547401611", 
"https://github.com" } } ]

jofemodo, 2019-10-29:
Hi  @Mickinator !

Please, the place for this kind of questions is the forum.

Thanks!

mheidt, 2019-11-10:
We said, that it is not not necessary until we found an instrument that sends non-0 values for both, MSB and LSB.

closed 2019-11-10 by mheidt