zynthian/zynthian-ui #171: Can't choose presets for certain lv2 synths

Baggypants, 2019-11-05:
**Describe the bug**
It appears that commit: 74e3ce5  in zynthian-ui has removed the ability to choose presets for obxd, MDA ePiano and some other lv2 synths.

**To Reproduce**
Steps to reproduce the behavior:
1. new synth
2. obxd
3. choose midi ch
4. get taken directly to controls, no presets available.

**Expected behavior**
see presets

**Setup:**
 - SD-image: buster
 - Hardware Config: V2
 - Software Version

zyncoder: master (8b7bfbd)
zynthian-ui: master (95e8a9d)
zynthian-sys: master (2ba5598)
zynthian-data: master (402a86a)
zynthian-webconf: master (dfb672f)

jofemodo, 2019-11-05:
It should be fixed now. Please, update & test.

closed 2019-11-05 by jofemodo

Baggypants, 2019-11-05:
Tested yc-20 and obxd. Works correctly. You can see the presets in obxd and it takes you directly the the controls when there are none.

commit 8748619c (2019-11-05): Fix #171: Some plugins doesn't allow to select presets.