zynthian/zynthian-ui #172: Implement CUIA toggle record function.

wyleu, 2019-11-13:
**Is your feature request related to a problem? Please describe.**
In an environment using foot switches to operate Recording/Stop it would be preferable to operate this on one switch.

**Describe the solution you'd like**
The ability to define a specific device switch (or input event) to toggle the record status. 

**Describe alternatives you've considered**
A short press for Record and a long press for top record implements on the I/O pins but this requires implementation of the electrical contacts involved with associated connector choices, and construction.
The USB implementation requires no assembly. 

[image]

jofemodo, 2019-11-13:
It's merged in PR #173

jofemodo, 2019-11-13:
Also documented in wiki ;-)

closed 2019-11-13 by jofemodo