zynthian/zynthian-ui #175: Implement an "ALSA Mixer" panel

jofemodo, 2019-11-21:
It could be implemented as "Engine", a kind of "effect" where the controls are the mixer levels.

jofemodo, 2019-11-21:
It should be remotely accesible by OSC (webconf) and levels should be saved with snapshots.

riban-bw, 2019-11-21:
This us similar to the idea we were talking about earlier this year of having an effects chain after the engines, like an insert point in a mixer. Also a master mix bus with an effcts chain. With such points in the audio path we can add effects, each with OSC / MIDI control.

jofemodo, 2019-12-03:
Checkout "mixer" branch ;-)

jofemodo, 2019-12-19:
MIDI control is very slow because the use of amixer as interface for the ALSA mixer. We should study to use a faster interface.

jofemodo, 2019-12-20:
A sender poll thread has been implemented for avoiding the "slow" MIDI control problem. Now it works like a charm ...

closed 2019-12-20 by jofemodo