zynthian/zynthian-ui #176: Feature request - GUI for Helm

Humihumihumi, 2019-11-26:
Please implement a GUI interface for Helm.

Thanks. 

[image]

jofemodo, 2019-11-26:
We should find a way of getting the native display of any plugin LV2.

riban-bw, 2019-11-27:
LV2 provides a mechanism to provide GUI for each plugin. The GUI acts like any other client and accesses the plugin via its ports. Only parameters exposed by control ports may be controlled hence the level of control is the same as (or less than) that provided by Zynthian. (Some ports may be hidden from the GUI.) Not all LV2 plugins provide GUIs. JALV provides some hosts for GUIs based on GTK and QT, e.g. jalv.gtk, jalv.qt5, etc. These will display the LV2 gui if it exists otherwise present a set of control knobs. Some plugins do have GUIs but fail to display due to bugs or configuration issues. Most of the GUIs are of little use and don't look great. b_whirl#extended is quite nice. There is no LV2 GUI for Helm.

Many LV2 plugins will run on the Zynthian with jalv.gtk but some fail. We could have a table of those that will run with a useful GUI then launch just those with jalv.gtk but I think most people will be disappointed because they really want the full applications (like Helm) which give functionality not exposed in the LV2 plugin controls.

riban-bw, 2021-03-18:
LV2 with working GUI can now be displayed via VNC. Some still fail. This issue is captured in zynthian/zynthian-ui/issues/352.
Helm is now working so will close this issue.

closed 2021-03-18 by riban-bw