zynthian/zynthian-ui #177: Midi In Hickup with NoteOff/Vel0 and new ttymidi

mheidt, 2019-12-01:
Setup:
V2 Kit
D50 over ttymidi
Buster image

I am playing trills over 59/60 and it works when I use a non-buster image.
In those old images I see the D50 sending NoteOn Vel0.

But in buster they became NOTE-OFFs.
Apparently because of the new ttymidi change mentioned earlier in the forum.
Unfortunately there are hickups.

CH#00 CONTROL_CHANGE 123 => 0
CH#00 NOTE_OFF 59, Vel: 64
CH#00 NOTE_ON 73, Vel: 59
CH#00 NOTE_OFF 123, Vel: 64
CH#00 NOTE_ON 60, Vel: 88
CH#00 NOTE_OFF 60, Vel: 64

I can understand the OFF 123....a bug with the value of the CONTROL_CHANGE...but I can't explain the 73.

I could confirm this behaviour on a second v1 MCP23008 zynthian as well.
Works well with pre buster and fails with buster.

riban-bw, 2019-12-02:
Does the D50 send messages with running status? I wonder whether the changes to ttyMidi properly support this feature.

jofemodo, 2019-12-02:
It looks like something is not working OK with the mod-ttymidi running status implementation.
Please, test with older versions and try to figure out the commit that breaks things. When it's clear where the problem is, i would roll-back the zynthian version and open an issue in mod-ttymidi repo.

Regards,