zynthian/zynthian-ui #179: Midi-Recorder needs bpm-value-window

Mickinator, 2019-12-07:
If I play my drumpatterns in loop mode it would be good to have a bpm-poti to play the rhythm faster or slower

jofemodo, 2019-12-09:
I've been testing jack-smf-player with jack transport, changing the BPMs, but it doesn't work. Reading the jack-smf-player code, i found this nice comment:

```
static int
sync_callback(jack_transport_state_t state, jack_position_t *position, void *notused)
{
	assert(jack_client);

	/* XXX: We should probably adapt to external tempo changes. */

	if (state == JackTransportStarting) {
		song_position = position->frame;
		smf_seek_to_seconds(smf, nframes_to_seconds(position->frame));

		if (!be_quiet)
			g_debug("Seeking to %f seconds.", nframes_to_seconds(position->frame));

		playback_started = jack_frame_time(jack_client);

	} else if (state == JackTransportStopped) {
		playback_started = -1;
	}

	return TRUE;
}
```

So, it look like we have some work to do ... ;-)

Regards,

jofemodo, 2019-12-09:
Ahh! For those interested in helping with this, jack-smf-utils code is here:

https://github.com/zynthian/jack-smf-utils

Mickinator, 2019-12-09:
> <mailto:notifications@github.com> >
> (#179)
> <mailto:zynthian-ui@noreply.github.com> >
> <mailto:norbert.schreiber@t-online.de> >, Author <
> author@noreply.github.com <mailto:author@noreply.github.com> >

👍🌈

von Norbert Schreiber gesendet

jofemodo, 2019-12-09:
OK! jack-smf-player now support BPM changes:

https://github.com/zynthian/jack-smf-utils/issues/1

We can implement the tempo control in the MIDI recorder. It's a big step forward because zynthian will have a tempo line with transport control ... and this is the basic infrastructure for many new features ;-)

Mickinator, 2019-12-10:
Wow, Wow, wow,
I can't test it now, but tomorrow in the morning 😂😂😂🌈
You're great, so great, thank you very much !!!

jofemodo, 2019-12-10:
Nothing to test yet, my friend! I only implemented the feature in the "core" level. Now i have to implement it at the user level (UI).

Mickinator, 2019-12-10:
Ok, I will wait. But I'm so glad that I know you and your team and 
Zynthian grows (nearly every day).
I can't wait till tomorrow. But now I have to go to bed. I believe, I can't 
sleep 😂😂😂😂😂
Thanx, thanx, thanx

Gesendet mit der Telekom Mail App
<http://www.t-online.de/service/redir/emailmobilapp_ios_smartphone_footerlink.htm>

Datum: 10.12.2019, 01:27 Uhr

jofemodo, 2019-12-12:
You can test the new feature in the "zyntransport" branch.

Mickinator, 2019-12-12:
this night 👍😂🌈

von Norbert Schreiber gesendet

Datum: 12.12.2019, 13:31 Uhr

Mickinator, 2019-12-13:
**Jofemodo:** You can test the new feature in the "zyntransport" branch.
**Mickinator:** Oh,I can´t find the "zyntransport branch"

Is there anybody who can help?

mheidt, 2019-12-13:
have you updated your zynthian?

Mickinator, 2019-12-13:
yes, several times

Gesendet mit der Telekom Mail App
<http://www.t-online.de/service/redir/emailmobilapp_ios_smartphone_footerlink.htm>

(#179)

Mickinator, 2019-12-13:
I use the buster image.

Gesendet mit der Telekom Mail App
<http://www.t-online.de/service/redir/emailmobilapp_ios_smartphone_footerlink.htm>

(#179)

jofemodo, 2019-12-13:
You have to change the branch in webconf->repositories.
And update software ...

Mickinator, 2019-12-14:
I´ve changed Software/repositories/Zynthian-ui to zyntransport .
zynthian-webconf is on master.
Then i made an update -> Zynthian Display says "Error"
Then I made a roboot -> Zynthian Display says "Error"

After changing Zynthian-ui to master I could work again,but without bpm-cintrolling.

Do I have to change anything  in zynthian-webconf too?

jofemodo, 2019-12-14:
Yes. Please go to Hardware->Audio, click save and reboot.
I will try to find a way of including this in the update ...

Mickinator, 2019-12-15:
Hi Fernando,
now I have it. A lot of thanks.
But there is one problem: I recordet my drumpatterns with 80 bpm or 120 bpm. But if I playback my patterns, the speed is very low. With 360 bpm shown in the Zynthian-Display, it reaches max 60 bpm. I can´t go faster because 360 is the maximum. ...

jofemodo, 2019-12-15:
Please, send me one of your MIDI tracks. 

Thanks!

Mickinator, 2019-12-16:
Here is a track 
[Midi Track.zip](https://github.com/zynthian/zynthian-ui/files/3967542/Midi.Track.zip)

and here are 2 Examples:
the 1st is a normal pattern with zynthian-ui= master
and the 2nd is the speeddown-example beginning with 360 the 110 and the 360 again
[Zynthian.zip](https://github.com/zynthian/zynthian-ui/files/3967574/Zynthian.zip)

jofemodo, 2019-12-17:
It should be solved now ...

Regarding the "retrigger" problem, i can't reproduce it, sorry ...

Mickinator, 2019-12-17:
Hi Fernando,

yes, the bpm-controö works now. Thanks! This helps a lot :-)

Regarding the retrigger I´ve made a short video. Please excuse, my smartphone always stops recording automaticly ... I have more then 1 GB free space, but ....

The video shows, that when I´m using zynthian-ui: zyntransport (4fdea8d) the red frame jumps two rows up or down by pushing the up-down or select short-botton.

If I use the zynthian-ui:master everything works perfectly, but I have no speed-control (sorry, no vid)

[dobbletrig.zip](https://github.com/zynthian/zynthian-ui/files/3972953/dobbletrig.zip)

jofemodo, 2019-12-17:
Sorry, i can't reproduce your problem. MIDI control works perfectly for me. If i can't reproduce, i can't fix.

Regarding the first event problem, i've noted the "effect" and will try to solve it ... 
The error when stopping should be solved. Try updating again ...

Mickinator, 2019-12-17:
doubble trigger: bad luck, but things happend. I kow you give your very best. And I can´t solve it. So, lot of thanks. I believe the note-commands go two different ways to the midi-output. But I will controll this steps via Qwert-Keyboard.

Thanks a lot for solving the other probs.

Mickinator, 2019-12-18:
doubble trigger is solved, works now perfectly
Speed Control is ok.
noisy Sound by starting and stopping is not solved.

But: It goes foreward step by step. You are a magician in programming!
Thank you so much !!!

Mickinator, 2019-12-19:
Sorry, new problem
The Midi recorder drops the last tick of a trippled 4/4 track.
In the zip are wavs what show plaback via Zynthian and playback via PC.

[Sidestick.zip](https://github.com/zynthian/zynthian-ui/files/3983519/Sidestick.zip)

jofemodo, 2019-12-20:
Hi @Mickinator !
I'm closing this. Please, open a new issue for any related problem ...

commit eaca7389 (2019-11-13): Fixes #179: Pianoteq version is not correctly detected and config file is generated with wrong name.
closed 2019-12-20 by jofemodo