zynthian/zynthian-ui #187: Midi player not playing correct bpm

mheidt, 2019-12-22:
The attached midi file is playing 4th at C4 for 2 minutes.
At least timidity would do so.

INFO:zynthian_gui_midi_recorder: STARTING MIDI PLAY '/zynthian/zynthian-my-data/capture/C4th.mid' ...
INFO:zynthian_gui_midi_recorder: COMMAND: /usr/local/bin/jack-smf-player -n -s -a "ZynMidiRouter:seq_in" -r 63 "/zynthian/zynthian-my-data/capture/C4th.mid"
DEBUG:zynthian_gui_controller: ZCTRL 'BPM': 120 (10 -> 400), None, None
DEBUG:zynthian_gui_midi_recorder: SET PLAYING BPM => 120
DEBUG:zynthian_gui_midi_recorder: SET PLAYING BPM => 120
DEBUG:zynthian_gui_controller: ZCTRL 'BPM': 120 (10 -> 400), None, None
DEBUG:zynthian_gui_midi_recorder: SET PLAYING BPM => 120
DEBUG:zynthian_gui_controller: ZCTRL 'MIDI Recorder': 0 (0 -> 6), None, None

Even when I increase to 400 BPM, it's not at 120

It's tested on a 2019-11-29 (updated)

[C4th.mid.zip](https://github.com/zynthian/zynthian-ui/files/3993127/C4th.mid.zip)

jofemodo, 2019-12-23:
Why channel 11?  Is the problem affecting all channels or only channel 11?

mheidt, 2019-12-23:
My USB is programmed on channel 11. 
It is a LinuxSampler issue. When I create a ZY layer, it works. Maybe it is only relates to the sleigh bell sound. haven't checked.

jofemodo, 2020-02-11:
Hi @mheidt !

Do you have extra info about this issue?

Regards,

mheidt, 2020-02-11:
[sleigh_bells.zip](https://github.com/zynthian/zynthian-ui/files/4188804/sleigh_bells.zip)
Upload the soundfont into the sf2 folder and the mid into the captures.
Load the snapshot on Midi Channel 11.
Run the mid file in a loop. 
You should hear the sleighbells but very slowly. When you crank up to 300bpm, it's near a quarter beat.

mheidt, 2020-02-11:
maybe the channel 11 wasn't necessary and I created the mid file on channel1 but was running in single layer mode

riban-bw, 2021-07-05:
MIDI playback is now performed by bespoke Zynthian library. Please retest.

closed 2021-11-30 by mheidt