zynthian/zynthian-ui #191: AudioInjector Ultra / Mixer

mheidt, 2019-12-31:
DEBUG:zynthian_engine_mixer: MIXER CONTROL => DAC
'DAC',0
  Capabilities: volume pswitch pswitch-joined
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 255
  Front Left: 245 [82%] [-5.00dB] Playback [off]
  Front Right: 245 [82%] [-5.00dB] Playback [off]
DEBUG:zynthian_engine_mixer:   Capabilities => volume pswitch pswitch-joined
DEBUG:zynthian_engine_mixer:   Playback channels => Front Left - Front Right
DEBUG:zynthian_engine_mixer:   Capture channels => Front Left - Front Right
DEBUG:zynthian_engine_mixer:   Limits => 0 - 255
DEBUG:zynthian_engine_mixer:   Front Left => 245 [82%] [-5.00dB] Playback [off]
DEBUG:zynthian_engine_mixer:   Front Right => 245 [82%] [-5.00dB] Playback [off]

This one is shown, but with a datavalue-max of 1

and the input aka PGA is not shown at all
DEBUG:zynthian_engine_mixer: MIXER CONTROL => PGA
'PGA',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 48
  Front Left: 38 [79%] [7.00dB]
  Front Right: 38 [79%] [7.00dB]

DEBUG:zynthian_engine_mixer:   Capabilities => volume
DEBUG:zynthian_engine_mixer:   Playback channels => Front Left - Front Right
DEBUG:zynthian_engine_mixer:   Capture channels => Front Left - Front Right
DEBUG:zynthian_engine_mixer:   Limits => 0 - 48
DEBUG:zynthian_engine_mixer:   Front Left => 38 [79%] [7.00dB]
DEBUG:zynthian_engine_mixer:   Front Right => 38 [79%] [7.00dB]

mheidt, 2020-01-01:
In alsamixer DAC and PGA show up as Playback and Capture.
I would just add the following lines

						elif 'volume' in ctrl_caps:
							ctrl_type = "Playback"
I cannot control the playback and capture channels separately

mheidt, 2020-01-10:
Could you please add this line to the file? I don't want to create a branch just for this line...

jofemodo, 2020-01-12:
Done!

commit 53c2e1ca (2020-01-12): Fix #191: Add "volume" to supported capabilities in mixer (by @mheidt).
closed 2020-01-12 by jofemodo