zynthian/zynthian-ui #193: zynthian_engine_mixer runs invalid amixer command

jypma, 2020-01-12:
I'm using a USB Behringer UMC404HD as sound card, and selected the default settings for it. When coming up, zynthian writes this error to the log:
```
zynthian startx[405]: ERROR:zynthian_engine_mixer: Command 'amixer -M -c CARD=U192k' returned non-zero exit status 1.
```
and sure enough, you can't pass `CARD=U192k` as `-c` to `amixer`:
```
root@zynthian:~# amixer -M -c CARD=U192k
Invalid card number.
Usage: amixer <options> [command]
[...]
```
I'm not sure which logical option to pass instead, but a simple `-c 1` does work. I guess that would make it order-dependent though...
```
root@zynthian:~# amixer -M -c 1
Simple mixer control 'Mic',0
  Capabilities: cvolume cswitch
  Capture channels: Front Left - Front Right
  Limits: Capture 0 - 127
  Front Left: Capture 127 [100%] [0.00dB] [on]
  Front Right: Capture 127 [100%] [0.00dB] [on]
Simple mixer control 'Mic',1
  Capabilities: cvolume cvolume-joined cswitch cswitch-joined
  Capture channels: Mono
  Limits: Capture 0 - 127
  Mono: Capture 127 [100%] [0.00dB] [on]
Simple mixer control 'UMC204HD 192k Output',0
  Capabilities: pvolume pswitch
  Playback channels: Front Left - Front Right - Rear Left - Rear Right
  Limits: Playback 0 - 127
  Mono:
  Front Left: Playback 127 [100%] [0.00dB] [on]
  Front Right: Playback 127 [100%] [0.00dB] [on]
  Rear Left: Playback 127 [100%] [0.00dB] [on]
  Rear Right: Playback 127 [100%] [0.00dB] [on]
Simple mixer control 'UMC204HD 192k Output',1
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback 0 - 127
  Mono: Playback 127 [100%] [0.00dB] [on]
```

mheidt, 2020-01-12:
what does aplay -l say?

this is mine:
card 0: audioinjectorul [audioinjector-ultra],

And my jackd parameters are
-P 70 -t 2000 -s -d alsa -d hw:audioinjectorul -r 48000 -p 256 -n 2 -X raw

whatever shows after card0: and before the brackets, should be used in hw:

mheidt, 2020-01-12:
It was you, who submitted this card as hw:CARD=U192k, btw.

jofemodo, 2020-01-12:
Please, test removing the "CARD=" part. From webconf->hardware->audio, change jackd options like this:

`-P 70 -t 2000 -d alsa -d hw:U192k -r 48000 -p 256 -n 3 -s -S -X raw
`

Reboot and test.

Thanks!