zynthian/zynthian-ui #194: Input is connected to output when FX Layer is active

jypma, 2020-01-12:
On a clean start and no snapshot loaded (USB Behringer UMC404HD), jack/zynthian connects input 1 and 2 to output 1, apparently:
```
root@zynthian:~# jack_lsp -c
system:capture_1
   system:playback_1
system:capture_2
   system:playback_1
[...]
```
This seems unrelated to the hardware monitoring switches on the sound card itself, since they always work regardless of jack settings (hence, "hardware" monitoring :-).

Now adding an FX Layer will add the effects _additional_ to the already routed-through input, giving a double signal.

Now a simple `jack_disconnect system:capture_1 system:playback_1` will get rid of it, but I'm wondering what might be causing it? Potentially it's related to the new mixer layer not yet working for my sound card (see #193), but I don't think it is.

jypma, 2020-01-12:
Update: if I manually disconnect them using `jack_disconnect`, they'll actually get reconnected when I add a new FX Layer. That's strange indeed.

jofemodo, 2020-01-12:
Please, update your zynthian and try ...

jypma, 2020-01-13:
Thanks! I'll give it a spin tonight.

jypma, 2020-01-13:
Unfortunately, that didn't seem to do the trick. The input to output connections are different than before, though.

After clean boot:
```
root@zynthian:~# jack_lsp -c
system:capture_1
   system:playback_1
system:capture_2
   system:playback_2
system:playback_1
   system:capture_1
system:playback_2
   system:capture_2
```
After hooking up an effect:
```
root@zynthian:~# jack_lsp -c
system:capture_1
   system:playback_1
   C* CabinetIII - Idealised loudspeaker cabinet emulation:in
system:capture_2
   system:playback_2
   C* CabinetIII - Idealised loudspeaker cabinet emulation:in
system:playback_1
   system:capture_1
   C* CabinetIII - Idealised loudspeaker cabinet emulation:out
system:playback_2
   system:capture_2
   C* CabinetIII - Idealised loudspeaker cabinet emulation:out
```
I'm running off e4907becbac23140f5c8644f6f8994b89b66169e :
```
root@zynthian:/zynthian/zynthian-ui# git log -n 1
commit e4907becbac23140f5c8644f6f8994b89b66169e (HEAD -> master, origin/master, origin/HEAD)
Author: jofemodo <fernando@zynthian.org>
Date:   Sun Jan 12 22:57:10 2020 +0100

    Input is connected to output when FX Layer is active. Fix #194
```

jypma, 2020-01-13:
I wonder, what does `if rl.jackname` do?

I just spotted the `if zynthian_gui_config.midi_aubionotes_enabled` line below. My "Enable AubioNotes (Audio2MIDI)" is OFF in the webconf UI.

jypma, 2020-01-13:
I've been trying to take a poke at `zynthian_autoconnect.py`, but my `print`-debugging seems to be getting nowhere. Is it true that `print('foo')` doesn't show anywhere? It seems not to go into what's shown under `journalctl -u zynthian`, at least. 

How can I add some debug logging?

jypma, 2020-01-13:
OK, I've managed to trace it.

Even after a clean boot, it seems that there's always a "synth" layer, which is somehow seen, that makes `zynthian_autoconnect` connect input directly to output. I've traced it to [here](https://github.com/zynthian/zynthian-ui/blob/e4907becbac23140f5c8644f6f8994b89b66169e/zynautoconnect/zynthian_autoconnect.py#L334). If I comment out those lines, the erroneous extra input connection isn't made, and `system:capture` is correctly hooked up to the Effect.

However, that fixes only the input part... now the _output_ of the FX is no longer routed to `system:playback` anymore, either. 

Well, that's how far I'm going to get today. Hope it helps a little :-)

jofemodo, 2020-01-13:
OK! 
I can't test because i've not a zynthian with audio input here, but I think i catched the problem. Please, update and try now ...

jypma, 2020-01-14:
I had a quick test with a few of the C* plugins included with Zynthian.

first _C* Compress - Mono compressor_, working fine:
```
system:capture_1
   C* Compress - Mono compressor:in
system:capture_2
   C* Compress - Mono compressor:in
system:playback_1
   C* Compress - Mono compressor:out
system:playback_2
   C* Compress - Mono compressor:out
```
Then, trying _C* CompressX2 - Stereo compressor_, here only the left input channel gets used, but both output channels:
```
system:capture_1
   C* CompressX2 - Stereo compressor:in_l
system:capture_2
   C* CompressX2 - Stereo compressor:in_l
system:playback_1
   C* CompressX2 - Stereo compressor:out_l
system:playback_2
   C* CompressX2 - Stereo compressor:out_r
```
With _C* CabinetIII - Idealised loudspeaker cabinet_, all is fine, probably because it's mono too:
```
system:capture_1
   C* CabinetIII - Idealised loudspeaker cabinet emulation:in
system:capture_2
   C* CabinetIII - Idealised loudspeaker cabinet emulation:in
system:playback_1
   C* CabinetIII - Idealised loudspeaker cabinet emulation:out
system:playback_2
   C* CabinetIII - Idealised loudspeaker cabinet emulation:out
```
But surprisingly, _C* AmpVTS - Tube amp + Tone stack_ doesn't get hooked up at all:
```
system:capture_1
system:capture_2
system:playback_1
system:playback_2
[...]
C* AmpVTS - Tube amp + Tone stack:in
C* AmpVTS - Tube amp + Tone stack:out
```
So nothing is connected. I'm not sure what makes that last one so special; it looks like a normal mono plugin. Perhaps its LV2 ports are different somehow?

Either way, I suppose stereo plugins would need some different handling... either make the routing (more) stereo-aware, or only do mono -> mono and stereo -> stereo, perhaps requiring an explicit mono -> stereo plugin in the chain.

I didn't test any stacked effects.

closed 2020-01-15 by jofemodo

jofemodo, 2020-01-15:
It's solved now. The current behaviour is:

+ If the first effect-plugin in a layer (root) is mono, both audio-capture channels will be connected to the mono input, down-mixing the stereo audio-stream to mono.

+ If the first effect-plugin in a layer (root) is stereo, each audio-capture channel will be connected to one of the plugin's stereo audio inputs. If you only use stereo plugins, you have a stereo effect-chain.

Of course, this doesn't cover all use-cases. In fact, doing so would be difficult (although not impossible) to implement in the Zynthian UI. Anyway, we can solve some of the most typical use-cases, like:

+ Using a single audio input, but getting a "stereo" output. The simplest way of solving this is inserting a mono Gain (or any other mono plugin) as root effect.

+ Having a separated effect-chain for every input channel. Solving this would require some coding for implementing a mechanism that allows to configure per-layer audio channel: Stereo, Right, Left.

I've opened a new feature request for that:

https://github.com/zynthian/zynthian-ui/issues/195

Regards,

commit e4907bec (2020-01-12): Input is connected to output when FX Layer is active. Fix #194
commit 0e8f5be1 (2020-01-15): Fix #194: When connecting audio capture to a root layer (root effect plugin), use both channels if available.