zynthian/zynthian-ui #195: Per-layer audio-channels configuration: Stereo, Right, Left

jofemodo, 2020-01-15:
Currently, all layers are stereo. It would be desirable to have mono layers (Right or Left), so we can have 2 separated chain effects, or 2 synth layers with separated audio outputs, etc.

jypma, 2020-01-16:
Some ideas for UX (assuming 4 input channels and 4 output channels, for maximum complexity). Not all variants would be needed right away, I assume :-)

When a selection can be made (more input channels than FX inputs, or more playback channels than FX output), display the following menu(s).

## When selecting a mono FX as root
```
[CREATING MONO FX]
--- Select input: ---
Input (1)
Input (2)
Input (1 + 2)
Input (3)
Input (4)
Input (3 + 4)
```
where `(1 + 2)` would downmix into mono, just like right now.
## When selecting a stereo FX as root
```
[CREATING STEREO FX]
--- Select input: ---
Input (1 + 2)
Input (3 + 4)
```
where `(1 + 2)` means a normal stereo connection of those inputs to the FX.
## Adding a stereo FX to a mono chain
Hook the mono output up to both stereo inputs (we may be doing this already)
## Adding a mono FX to a stereo chain
Connect both stereo outputs to the mono input (we may be doing this already)
## When selecting a stereo FX as root, or when changing a mono FX chain to stereo:
```
[CREATING STEREO FX]
--- Select output: ---
Output (1 + 2)
Output (3 + 4)
```
## When selecting a mono FX as root, or when changing a stereo FX chain to mono
```
[CREATING MONO FX]
--- Select output: ---
Output (1)
Output (2)
Output (1 + 2)
Output (3)
Output (4)
Output (3 + 4)
```
where `(1 + 2)` would output the mono output to both playback channels.
This last option is for the common use case of having a mono guitar go into headphones outputs, which are stereo.

jofemodo, 2020-01-17:
I think the output configuration can be simplified to:

+ Mono
+ Stereo

Anyway, zynthian currently does not support more than 2 audio channels, and i prefer to keep it simple by now. Zynthian is not a DAW .... yet! ;-)

So configuration options for "audio input" would be:

+ Mono R
+ Mono L
+ Mono R+L => and we avoid the need for a "mono" root-plugin when wanting to do this.
+ Stereo

Regards,

jypma, 2020-01-17:
I totally get that, we'll wait for the "export Ardour channel strip configuration to Zynthian" feature another time ;-) 

I do expect most zynthians sticking with the 1- or 2-input Pi I2S soundcards. USB does work, but I find it draws a lot (more?) power from my battery-based zynthian setup. Perhaps CPU as well, I'll measure that when I have time to build my second zynthian to compare.

jofemodo, 2020-06-10:
Include this use-case if possible:

https://discourse.zynthian.org/t/using-multiple-sound-cards-with-alsa-in/4309

jypma, 2020-06-10:
Unrelated to the above, but a good current workaround for 2-input sound cards is the built-in "Calf Stereo Tools" plugin as first effect on a FX Chain. That will allow you to select only L, only R, or any other combination, and treat everything as stereo from there on downwards.

jofemodo, 2020-06-14:
I just committed "per audio-channel" audio output routing (for system audio output).  I mean, you can route the audio output from a layer to:

+ System:playback_1 (Mono Left)
+ System:playback_2 (Mono Right)
+ Both (Stereo)

I will do the same with Audio Input ASAP ... ;-)

Regards,

jypma, 2020-06-14:
@jofemodo I actually got started on the input part over the last few days, since I realized an earlier feature I was missing (NOT having any capture input going into a layer, [see discourse](https://discourse.zynthian.org/t/zs3-preset-should-leave-certain-controls-alone/4310/5)) would fit nicely into it.

I came up with a matrix-style configuration, see [this commit](https://github.com/zynthian/zynthian-ui/compare/master...jypma:input_config?expand=1). I built it with more than 2 capture inputs, and more than 2 LV2 inputs in mind:

- Displayed is a list of capture "outputs"
- Pressing [select] on any capture output will toggle which LV2 input(s) it goes to, in the following order: [1], [2], [1 and 2], [3], [4], [3 and 4], [none] (for 4 input; LV2 inputs are treated in pairs that way)

Not sure about the total direction, but I wanted to share it since you also seem to be working on it. Use it as you see fit :-)

jofemodo, 2020-06-14:
Nice! I will test it ...
Regarding development workflow, i think it's simpler if you create new branches in the "official" zynthian-ui repository, so we can test your changes easily by choosing the branch from the webconf tool.

jofemodo, 2020-06-14:
I merged your branch into the development branch "jypma_input_config" ...

jofemodo, 2020-06-19:
This feature is implemented in these 2 commits:

+ https://github.com/zynthian/zynthian-ui/commit/11e7f947685cca098dee59c71bf4bf6359d1422a
+ https://github.com/zynthian/zynthian-ui/commit/5943fb0bd216ff1e8a4c1a880389112e58c165d4

jofemodo, 2020-06-19:
Please @jypma and others, update and check the new per-layer capture configuration.

If somebody have a USB sound-card with more than 2 input/outputs, it would be interesting to know if routing/capture is working for all channels (input and output). It should  ...

riban-bw, 2020-06-29:
I can't seem to get this working. What is an example workflow to demonstrate its use?

jofemodo, 2020-06-29:
I can't reproduce this error, @riban-bw. I've checked and "ex_data_dir" var is defined here:

https://github.com/zynthian/zynthian-ui/blob/1c7d11b63e0c8fb8da459aff22775402722e5ae2/zyngine/zynthian_engine.py#L52
 
So i'm afraid that the problem is on your setup. Please, check that your zynthian-ui branch is correctly set and updated.

Regarding the use-case, the most typical use-case would be having 2 separated mono FX-chains:

+ Electric Guitar
+ Mic (voice)

Or having 2 separated mono-synths with FXs:

+ Dexed + FXs
+ Raffo + FXs

Regards,

riban-bw, 2020-07-18:
I can confirm that this works for soundcards with more than two inputs and outputs. Each new layer is routed from inputs 1&2 to outputs 1&2.

I wonder if that is the best behaviour. I find myself adding layers then moving them to spare input channels which is irksome and interferes with existing audio paths but I think this needs some wider user input / suggestion, preferably from experience. I think we should keep the behaviour of routing to a single stereo output as this is the most common workflow, i.e. most people will have a main output they want all their audio routed to by default. The question is would people with multiple inputs expect each to be used (by default) for each new layer or are more people going to be using just inputs 1 & 2? I suggest that if you have paid out for an 8 input device you may well want to use it as such!

closed 2020-07-20 by jofemodo