zynthian/zynthian-ui #196: Can't restore FX chain on reboot (but fine in manual run)

jypma, 2020-01-17:
Turning on zynthian makes systemd-ran `zynthian` try and restore the state. In the state is a single FX chain with 3 effects. This fails (see output later).

If the `zynthian` service is stopped, and then zynthian is started manually from the console, the state restores fine.

The following output is in the journal:
```
Jan 17 09:05:31 zynthian startx[409]: ERROR:zynthian_engine: Can't start engine Jalv/GxBlueAmp => <pexpect.pty_spawn.spawn object at 0x6d4c9430>
Jan 17 09:05:31 zynthian startx[409]: command: /usr/local/bin/jalv
Jan 17 09:05:31 zynthian startx[409]: args: ['/usr/local/bin/jalv', 'http://guitarix.sourceforge.net/plugins/gx_blueamp_#_blueamp_']
Jan 17 09:05:31 zynthian startx[409]: buffer (last 100 chars): b'_load_bundle(): note: New version 2070.0 found in <file:///zynthian/zynthian-plugins/lv2/b_whirl/>\r\n'
Jan 17 09:05:31 zynthian startx[409]: before (last 100 chars): b'_load_bundle(): note: New version 2070.0 found in <file:///zynthian/zynthian-plugins/lv2/b_whirl/>\r\n'
Jan 17 09:05:31 zynthian startx[409]: after: <class 'pexpect.exceptions.TIMEOUT'>
Jan 17 09:05:31 zynthian startx[409]: match: None
Jan 17 09:05:31 zynthian startx[409]: match_index: None
Jan 17 09:05:31 zynthian startx[409]: exitstatus: None
Jan 17 09:05:31 zynthian startx[409]: flag_eof: False
Jan 17 09:05:31 zynthian startx[409]: pid: 698
Jan 17 09:05:31 zynthian startx[409]: child_fd: 29
Jan 17 09:05:31 zynthian startx[409]: closed: False
Jan 17 09:05:31 zynthian startx[409]: timeout: 20
Jan 17 09:05:31 zynthian startx[409]: delimiter: <class 'pexpect.exceptions.EOF'>
Jan 17 09:05:31 zynthian startx[409]: logfile: None
Jan 17 09:05:31 zynthian startx[409]: logfile_read: None
Jan 17 09:05:31 zynthian startx[409]: logfile_send: None
Jan 17 09:05:31 zynthian startx[409]: maxread: 2000
Jan 17 09:05:31 zynthian startx[409]: ignorecase: False
Jan 17 09:05:31 zynthian startx[409]: searchwindowsize: None
Jan 17 09:05:31 zynthian startx[409]: delaybeforesend: 0
Jan 17 09:05:31 zynthian startx[409]: delayafterclose: 0.1
Jan 17 09:05:31 zynthian startx[409]: delayafterterminate: 0.1
Jan 17 09:05:31 zynthian startx[409]: searcher: searcher_re:
Jan 17 09:05:31 zynthian startx[409]:     0: re.compile(b'\n> ')
Jan 17 09:05:31 zynthian startx[409]: <pexpect.pty_spawn.spawn object at 0x6d4c9430>
Jan 17 09:05:31 zynthian startx[409]: command: /usr/local/bin/jalv
Jan 17 09:05:31 zynthian startx[409]: args: ['/usr/local/bin/jalv', 'http://guitarix.sourceforge.net/plugins/gx_blueamp_#_blueamp_']
Jan 17 09:05:31 zynthian startx[409]: buffer (last 100 chars): b'_load_bundle(): note: New version 2070.0 found in <file:///zynthian/zynthian-plugins/lv2/b_whirl/>\r\n'
Jan 17 09:05:31 zynthian startx[409]: before (last 100 chars): b'_load_bundle(): note: New version 2070.0 found in <file:///zynthian/zynthian-plugins/lv2/b_whirl/>\r\n'
Jan 17 09:05:31 zynthian startx[409]: after: <class 'pexpect.exceptions.TIMEOUT'>
Jan 17 09:05:31 zynthian startx[409]: match: None
Jan 17 09:05:31 zynthian startx[409]: match_index: None
Jan 17 09:05:31 zynthian startx[409]: exitstatus: None
Jan 17 09:05:31 zynthian startx[409]: flag_eof: False
Jan 17 09:05:31 zynthian startx[409]: pid: 698
Jan 17 09:05:31 zynthian startx[409]: child_fd: 29
Jan 17 09:05:31 zynthian startx[409]: closed: False
Jan 17 09:05:31 zynthian startx[409]: timeout: 20
Jan 17 09:05:31 zynthian startx[409]: delimiter: <class 'pexpect.exceptions.EOF'>
Jan 17 09:05:31 zynthian startx[409]: logfile: None
Jan 17 09:05:31 zynthian startx[409]: logfile_read: None
Jan 17 09:05:31 zynthian startx[409]: logfile_send: None
Jan 17 09:05:31 zynthian startx[409]: maxread: 2000
Jan 17 09:05:31 zynthian startx[409]: ignorecase: False
Jan 17 09:05:31 zynthian startx[409]: searchwindowsize: None
Jan 17 09:05:31 zynthian startx[409]: delaybeforesend: 0
Jan 17 09:05:31 zynthian startx[409]: delayafterclose: 0.1
Jan 17 09:05:31 zynthian startx[409]: delayafterterminate: 0.1
Jan 17 09:05:31 zynthian startx[409]: searcher: searcher_re:
Jan 17 09:05:31 zynthian startx[409]:     0: re.compile(b'\n> ')
Jan 17 09:05:31 zynthian startx[409]: INFO:zynthian_autoconnect: ZynAutoConnect: MIDI ...
Jan 17 09:05:31 zynthian startx[409]: INFO:zynthian_autoconnect: ZynAutoConnect: Escaped for Audio ...
Jan 17 09:05:33 zynthian startx[409]: INFO:zynthian_autoconnect: ZynAutoConnect: MIDI ...
Jan 17 09:05:33 zynthian startx[409]: ERROR:zynthian_autoconnect: ZynAutoConnect ERROR: 'NoneType' object has no attribute 'encode'
```
Curiously, stopping the systemd-ran service, and manually invoking `/zynthian/zynthian-sys/sbin/zynthian.sh`, works fine, with the following output (I added some timestamps):
```
Fri 17 Jan 2020 09:27:05 GMT:  
Fri 17 Jan 2020 09:27:05 GMT:  
Fri 17 Jan 2020 09:27:05 GMT:  X.Org X Server 1.20.4
Fri 17 Jan 2020 09:27:05 GMT:  X Protocol Version 11, Revision 0
Fri 17 Jan 2020 09:27:05 GMT:  Build Operating System: Linux 4.15.0-48-generic armv8l Raspbian
[... 235 lines cut ...]
Fri 17 Jan 2020 09:27:17 GMT:  DEBUG:zynthian_layer: BANK LIST => 
Fri 17 Jan 2020 09:27:17 GMT:  [('NoBank', None, 'NoBank', None)]
Fri 17 Jan 2020 09:27:17 GMT:  INFO:zynthian_autoconnect: ZynAutoConnect: Audio ...
Fri 17 Jan 2020 09:27:17 GMT:  DEBUG:zynthian_autoconnect: Autoconnecting Engine GxCreamMachine ...
Fri 17 Jan 2020 09:27:17 GMT:  DEBUG:zynthian_autoconnect:  => Connecting to GxBlueAmp
Fri 17 Jan 2020 09:27:17 GMT:  DEBUG:zynthian_autoconnect: Autoconnecting Engine GxBlueAmp ...
Fri 17 Jan 2020 09:27:17 GMT:  DEBUG:zynthian_autoconnect:  => Connecting to Tal-Reverb
Fri 17 Jan 2020 09:27:17 GMT:  DEBUG:zynthian_autoconnect: Autoconnecting Engine Tal-Reverb ...
Fri 17 Jan 2020 09:27:17 GMT:  DEBUG:zynthian_autoconnect:  => Connecting to system
Fri 17 Jan 2020 09:27:17 GMT:  DEBUG:zynthian_layer: BANK LIST => 
Fri 17 Jan 2020 09:27:17 GMT:  [('NoBank', None, 'NoBank', None)]
Fri 17 Jan 2020 09:27:17 GMT:  DEBUG:zynthian_layer: PRESET LIST => 
Fri 17 Jan 2020 09:27:17 GMT:  [('file:///zynthian/zynthian-plugins/lv2/gx_CreamMachine.lv2/default-preset', None, 'Default', ['NoBank'])]
Fri 17 Jan 2020 09:27:17 GMT:  DEBUG:zynthian_gui: ACTIVE CHAN: -1 => 0
Fri 17 Jan 2020 09:27:17 GMT:  INFO:zynthian_gui: MIDI SWITCHES SETUP...
Fri 17 Jan 2020 09:27:17 GMT:  DEBUG:zynthian_gui_control: SET CONTROLLER SCREEN GxBlueAmp#1
Fri 17 Jan 2020 09:27:17 GMT:  DEBUG:zynthian_gui_controller: ZCTRL 'BYPASS': 1.0 (0.0 -> 1.0), None, None
Fri 17 Jan 2020 09:27:17 GMT:  DEBUG:zynthian_gui_controller: ZCTRL 'MASTER': 0.41732283464566927 (0.0 -> 1.0), None, None
Fri 17 Jan 2020 09:27:17 GMT:  INFO:zynthian_autoconnect: ZynAutoConnect: MIDI ...
Fri 17 Jan 2020 09:27:17 GMT:  INFO:zynthian_autoconnect: ZynAutoConnect: Escaped for Audio ...
Fri 17 Jan 2020 09:27:17 GMT:  DEBUG:zynthian_gui_controller: ZCTRL 'TONE': 0.6377952755905512 (0.0 -> 1.0), None, None
Fri 17 Jan 2020 09:27:17 GMT:  DEBUG:zynthian_gui_controller: ZCTRL 'VOLUME': 0.31496062992125984 (0.0 -> 1.0), None, None
Fri 17 Jan 2020 09:27:19 GMT:  INFO:zynthian_autoconnect: ZynAutoConnect: MIDI ...
Fri 17 Jan 2020 09:27:19 GMT:  INFO:zynthian_autoconnect: ZynAutoConnect: Escaped for Audio ...
```

jypma, 2020-01-17:
Update 1: `systemctl restart zynthian` does _not_ make it come up. But after that command, the Zynthian screen shows a white zynthian logo with red "Error", rather than the blank screen it showed on boot.

Update 2: `reboot` _does_ makes zynthian come up fine on the next boot... it seems to only happen on a "cold" boot. Repeatedly. Strange :)

Please let me know if you'd like me to try anything else.

I wonder, would it make sense to have one of the errors logged in my first message make the zynthian process kill itself, so `systemd` would "just" restart it? It would at least work around this problem :-)

jypma, 2020-01-18:
Update 3: running `top` during such a failing first boot shows `unattended-upgrades` taking a fair bit of CPU for a while (10 seconds or so). While it does stop afterwards, perhaps it interferes with the FX processes coming up properly?

This probably explains subsequent boots being succesful; `unattended-upgrades` probably only runs daily.

jypma, 2020-01-18:
I'm trying commenting out everything in `/etc/apt/apt.conf.d/20auto-upgrades`, to see if tomorrow's first boot will be any better :-)

jofemodo, 2020-01-20:
First, running zynthian.sh is not recommended with the latest versions, because the "Ctrl-C" won't stop all the processes. If you want to debug, then stop the "zynthian" service and start "zynthian_debug" service. You can see the logs with journalctl. Like this:
```

systemctl stop zynthian
systemctl start zynthian_debug
journalctl -f -u zynthian_debug
```
regarding the "unattended-upgrades", it's really curious, because it's explicitly disabled on the setup script:

`systemctl disable unattended-upgrades`

I've to do some research ...

Regards,

jypma, 2020-01-20:
Well this is strange
```
root@zynthian:~# systemctl status unattended-upgrades.service
● unattended-upgrades.service - Unattended Upgrades Shutdown
   Loaded: loaded (/lib/systemd/system/unattended-upgrades.service; disabled; vendor preset: enabled)
   Active: inactive (dead)
     Docs: man:unattended-upgrade(8)
```
so the systemd unit is not active for me, either. Yet, uncommenting those lines did seem to help: I turned my zynthian on today, and FX restored with no problem.

Debian says that `unattended-upgrades` is actually ran from `apt-daily`, which _IS_ active:
```
root@zynthian:~# systemctl status apt-daily.timer
● apt-daily.timer - Daily apt download activities
   Loaded: loaded (/lib/systemd/system/apt-daily.timer; enabled; vendor preset: enabled)
   Active: active (waiting) since Sat 2020-01-18 20:56:21 GMT; 1 day 18h ago
  Trigger: Mon 2020-01-20 18:05:48 GMT; 2h 44min left

Jan 18 20:56:21 zynthian systemd[1]: Started Daily apt download activities.
```

jypma, 2020-01-21:
Well, the `unattended-upgrades` may have been a red herring: Zynthian again didn't come up right today. I wasn't running `top` to see what else may have been running at the time.
```
Jan 21 17:25:55 zynthian startx[406]: ERROR:zynthian_engine: Can't start engine Jalv/GxCreamMachine => <pexpect.pty_spawn.spawn object at 0x6d451830>
Jan 21 17:25:55 zynthian startx[406]: command: /usr/local/bin/jalv
Jan 21 17:25:55 zynthian startx[406]: args: ['/usr/local/bin/jalv', 'http://guitarix.sourceforge.net/plugins/gx_CreamMachine_#_CreamMachine_']
Jan 21 17:25:55 zynthian startx[406]: buffer (last 100 chars): b'le(): error: Error reading file:///zynthian/zynthian-my-data/presets/lv2/manifest.ttl/manifest.ttl\r\n'
Jan 21 17:25:55 zynthian startx[406]: before (last 100 chars): b'le(): error: Error reading file:///zynthian/zynthian-my-data/presets/lv2/manifest.ttl/manifest.ttl\r\n'
Jan 21 17:25:55 zynthian startx[406]: after: <class 'pexpect.exceptions.TIMEOUT'>
Jan 21 17:25:55 zynthian startx[406]: match: None
Jan 21 17:25:55 zynthian startx[406]: match_index: None
Jan 21 17:25:55 zynthian startx[406]: exitstatus: None
Jan 21 17:25:55 zynthian startx[406]: flag_eof: False
Jan 21 17:25:55 zynthian startx[406]: pid: 685
Jan 21 17:25:55 zynthian startx[406]: child_fd: 28
Jan 21 17:25:55 zynthian startx[406]: closed: False
Jan 21 17:25:55 zynthian startx[406]: timeout: 20
Jan 21 17:25:55 zynthian startx[406]: delimiter: <class 'pexpect.exceptions.EOF'>
Jan 21 17:25:55 zynthian startx[406]: logfile: None
Jan 21 17:25:55 zynthian startx[406]: logfile_read: None
Jan 21 17:25:55 zynthian startx[406]: logfile_send: None
Jan 21 17:25:55 zynthian startx[406]: maxread: 2000
Jan 21 17:25:55 zynthian startx[406]: ignorecase: False
Jan 21 17:25:55 zynthian startx[406]: searchwindowsize: None
Jan 21 17:25:55 zynthian startx[406]: delaybeforesend: 0
Jan 21 17:25:55 zynthian startx[406]: delayafterclose: 0.1
Jan 21 17:25:55 zynthian startx[406]: delayafterterminate: 0.1
Jan 21 17:25:55 zynthian startx[406]: searcher: searcher_re:
Jan 21 17:25:55 zynthian startx[406]:     0: re.compile(b'\n> ')
Jan 21 17:25:55 zynthian startx[406]: <pexpect.pty_spawn.spawn object at 0x6d451830>
Jan 21 17:25:55 zynthian startx[406]: command: /usr/local/bin/jalv
Jan 21 17:25:55 zynthian startx[406]: args: ['/usr/local/bin/jalv', 'http://guitarix.sourceforge.net/plugins/gx_CreamMachine_#_CreamMachine_']
Jan 21 17:25:55 zynthian startx[406]: buffer (last 100 chars): b'le(): error: Error reading file:///zynthian/zynthian-my-data/presets/lv2/manifest.ttl/manifest.ttl\r\n'
Jan 21 17:25:55 zynthian startx[406]: before (last 100 chars): b'le(): error: Error reading file:///zynthian/zynthian-my-data/presets/lv2/manifest.ttl/manifest.ttl\r\n'
Jan 21 17:25:55 zynthian startx[406]: after: <class 'pexpect.exceptions.TIMEOUT'>
Jan 21 17:25:55 zynthian startx[406]: match: None
Jan 21 17:25:55 zynthian startx[406]: match_index: None
Jan 21 17:25:55 zynthian startx[406]: exitstatus: None
Jan 21 17:25:55 zynthian startx[406]: flag_eof: False
Jan 21 17:25:55 zynthian startx[406]: pid: 685
Jan 21 17:25:55 zynthian startx[406]: child_fd: 28
Jan 21 17:25:55 zynthian startx[406]: closed: False
Jan 21 17:25:55 zynthian startx[406]: timeout: 20
Jan 21 17:25:55 zynthian startx[406]: delimiter: <class 'pexpect.exceptions.EOF'>
Jan 21 17:25:55 zynthian startx[406]: logfile: None
Jan 21 17:25:55 zynthian startx[406]: logfile_read: None
Jan 21 17:25:55 zynthian startx[406]: logfile_send: None
Jan 21 17:25:55 zynthian startx[406]: maxread: 2000
Jan 21 17:25:55 zynthian startx[406]: ignorecase: False
Jan 21 17:25:55 zynthian startx[406]: searchwindowsize: None
Jan 21 17:25:55 zynthian startx[406]: delaybeforesend: 0
Jan 21 17:25:55 zynthian startx[406]: delayafterclose: 0.1
Jan 21 17:25:55 zynthian startx[406]: delayafterterminate: 0.1
Jan 21 17:25:55 zynthian startx[406]: searcher: searcher_re:
Jan 21 17:25:55 zynthian startx[406]:     0: re.compile(b'\n> ')
Jan 21 17:25:56 zynthian startx[406]: INFO:zynthian_autoconnect: ZynAutoConnect: MIDI ...
Jan 21 17:25:56 zynthian startx[406]: INFO:zynthian_autoconnect: ZynAutoConnect: Escaped for Audio ...
Jan 21 17:25:58 zynthian startx[406]: INFO:zynthian_autoconnect: ZynAutoConnect: MIDI ...
Jan 21 17:25:58 zynthian startx[406]: ERROR:zynthian_autoconnect: ZynAutoConnect ERROR: 'NoneType' object has no attribute 'encode'
```
A `reboot` took care of it.

jypma, 2020-01-21:
I wonder, can we have an option to _not_ try to restore the state on boot? With the variety of zynthian sounds, a quick boot might be preferred by some, just selecting a preset afterwards; plus, it at least would work around this issue :-)

jofemodo, 2020-01-21:
Have you tried the webconf's UI options? ;-)
There is a checkbox there ...  Take a look to the user's guide :

https://wiki.zynthian.org/index.php/Zynthian_UI_Users_Guide#Restore_Last_State

Regards,

jypma, 2020-01-21:
Thanks :-) I'm a whole order of magnitude more blind today. :coffee:

jofemodo, 2020-01-22:
I've disabled apt-daily.timer on future image builds.
You can disable it on your image by running:

`systemctl disable apt-daily.timer`

Anyway, i think the problem is not this (or at less, not only this) ...

jofemodo, 2020-01-22:
I see this message:

`Error reading file:///zynthian/zynthian-my-data/presets/lv2/manifest.ttl/manifest.ttl`

And it's curious ...

+ Have you upload LV2 presets by hand or using the webconf's preset manager?
+ Could you check if files:
    /zynthian/zynthian-my-data/presets/lv2/manifest.ttl/manifest.ttl
    or
    /zynthian/zynthian-my-data/presets/lv2/manifest.ttl/manifest.ttl

do exist? If they do, send me a copy and try removing them.

Regards,

jofemodo, 2020-01-22:
Running Buster, true? It's updated to the last?
Could you update and delete these files?

jypma, 2020-02-06:
A little time later, a little update. I've since done an `apt-get upgrade`, no changes. First boot of the day still fails with 
```
Feb 06 11:47:44 zynthian startx[403]: ERROR:zynthian_engine: Can't start engine Jalv/GxCreamMachine => <pexpect.pty_spawn.spawn object at 0x6d3ce710>
Feb 06 11:47:44 zynthian startx[403]: command: /usr/local/bin/jalv
Feb 06 11:47:44 zynthian startx[403]: args: ['/usr/local/bin/jalv', 'http://guitarix.sourceforge.net/plugins/gx_CreamMachine_#_CreamMachine_']
Feb 06 11:47:44 zynthian startx[403]: buffer (last 100 chars): b'_load_bundle(): note: New version 2070.0 found in <file:///zynthian/zynthian-plugins/lv2/b_whirl/>\r\n'
Feb 06 11:47:44 zynthian startx[403]: before (last 100 chars): b'_load_bundle(): note: New version 2070.0 found in <file:///zynthian/zynthian-plugins/lv2/b_whirl/>\r\n'
Feb 06 11:47:44 zynthian startx[403]: after: <class 'pexpect.exceptions.TIMEOUT'>
[...]
Feb 06 11:47:49 zynthian startx[403]: ERROR:zynthian_autoconnect: ZynAutoConnect ERROR: 'NoneType' object has no attribute 'encode'
```
I'm going to try the following, in order:
- [X] Disable `apt-daily.timer` and `apt-daily-upgrade.timer` _(no effect)_
- [ ] Remove all the .ttl preset files from `/zynthian/zynthian-my-data/presets/lv2`
- [ ] Disable "Restore state on startup"

jofemodo, 2020-02-09:
Have you tried with a really simple snapshot? Something like a single layer with ZynAddSubFX ...

regards

jypma, 2020-02-09:
Good tip! I'll try that.

Could you point me to the place where that `pexpect.exceptions.TIMEOUT` is set? I could try to raise the timeout (if that's it), to see it if makes it go past whatever ghost is waking up every day.

jofemodo, 2020-03-10:
This is probably solved now, in the latest nightly builds:

https://os.zynthian.org/2020-03-10-zynthianos-buster-lite-1.0.0.zip

or

https://os.zynthian.org/2020-03-09-zynthianos-buster-lite-1.0.0.zip

Please, confirm ...

jofemodo, 2020-03-31:
OK! I'm closing this ... re-open if you can reproduce it.

closed 2020-03-31 by jofemodo

jypma, 2020-04-01:
Can't reproduce after upgrading, let's consider this closed! Thanks a lot for the help.