zynthian/zynthian-ui #202: Deactivate SingleChannelMode when channels are cloned

mheidt, 2020-02-16:
I am not quite sure, if it is supposed to work, but cloning doesn't really work, if SingleChannelMode is activated.
I get hung notes.

Maybe we should just disable the SingleChannelMode, when layers are cloned (esp. after loading a snapshot)

jofemodo, 2020-02-17:
It works like a charm for me. Also when setting an arbitrary MIDI channel (let's say 11) on the keyboard.

Please, verify you are running last versions for UI and zyncoder library, as i made important changes the last week.

jofemodo, 2020-02-18:
Eooo! Could you verify this issue?

mheidt, 2020-02-18:
I have hanging nodes, but I can't confirm that the SingleChannelMode and cloned layers are the cause.
I will close and come back when I know better.

closed 2020-02-18 by mheidt