zynthian/zynthian-ui #207: Audio of a pure data layer is always routed to system whatever is the choice in audio routing

MrDham, 2020-03-04:
**Describe the bug**
Audio of a pure data layer is always routed to system whatever is the choice in audio routing

**To Reproduce**
Steps to reproduce the behavior:
Matérial: Zynthian, keyboard, audio system.  
1 Create a pure data layer, e.g. : 1#PD > synths/two ops1poly
2. Create an effect layer, e.g. : "2# > JV/MVerb > Stadium"
3. In main menu, bold click on "1#PD > synths/two ops1poly" and chose "audio routing"
4. Unselect "system" and play some notes.
5. Select "2# > JV/MVerb" and play some notes

**Expected behavior**
4. You should hear nothing
5. You should hear the sound of "1#PD > synths/two ops1poly" with a lot of reverb

** Problem ** 
4: You can hear the flat sound of "1#PD > synths/two ops1poly"
5. You can hear the flat sound of "1#PD > synths/two ops1poly" without a lot of reverb 

**Screenshots**
N/A

**Setup:**
 Matérial: Zynthian, keyboard, audio system.  
Buster RC1 + update at 03/03/2020
Kit V2
yncoder: master (c5a65aa)
zynthian-ui: master (a7ad5c6)
zynthian-sys: master (2655577)
zynthian-data: master (9cccef5)
zynthian-webconf: master (4a2f300)

jofemodo, 2020-03-04:
OK! It's solved now. Update & try!

closed 2020-03-04 by jofemodo

MrDham, 2020-03-04:
Fixed ! Thank you :-)

jofemodo, 2020-03-04:
I broke ZynAddSubFX by accident while solving this. Now everything should be working OK ...

commit 6048d090 (2020-03-04): Fix #207: PureData routing is failing due to bad jackname.