zynthian/zynthian-ui #210: Snapshots do not use the edited parameters

Mickinator, 2020-03-09:
Hello All, hallo Jose,

if I call a Snapshot using the clone function (clone ch# 1  to ch#2) eg Piano / Strings with Vol 100/25 and save that snapshot, the ch#2 volume is  much to high when I recall that snapshot.

Raspi 4
updated 09.03.20
ch1 = Pianotec Blüthner Vol 100
ch2 = Fluitsynth Strings Vol 25
clone 1 -> 2
save Snapshot

then call another sn and call the Piano/String-Sn again.
Every Time I have to edit the volume parameters again.

Thanks for helping 🌈

jofemodo, 2020-03-09:
Try now! It should be solved ...

closed 2020-03-09 by jofemodo

Mickinator, 2020-03-10:
> <mailto:notifications@github.com> >
> parameters (#210)
> <mailto:zynthian-ui@noreply.github.com> >
> <mailto:norbert.schreiber@t-online.de> >, Author <
> author@noreply.github.com <mailto:author@noreply.github.com> >

Hi Jóse,

thank you! It works. You're the best.
Btw. Is there a manual how to work with the mixer?

Have a nice week.
Regards
Norbert

Gesendet mit der Telekom Mail App
<https://www.telekom.de/unterwegs/apps-und-dienste/kommunikation/telekom-e-mail?wt_mc=alias_9998_unterwegs/apps-und-dienste/telekom-email#mail-app>

jofemodo, 2020-03-10:
All the documentation is on the Wiki:

https://wiki.zynthian.org

Regarding the Mixer, you can configure the parameters shown on the UI from the webconf:

**Hardware -> Audio -> Mixer Controls -> [+]**

And remember that this is a collaborative project, and you are invited to improve the wiki's documentation. If you want to do it, simply ask me for a wiki account ;-)

Regards,

commit 458ce529 (2020-03-09): Fix #210: Layer loading needed a "forced" call to MIDI autoconnector.