zynthian/zynthian-ui #211: MIDI Looping back on usb devices even though

1st-BrainStormer, 2020-03-10:
**Describe the bug**
USB Midi devices looping back.
**To Reproduce**
Steps to reproduce the behavior:
1. Plug in USB midi Keyboard (Casio CDP-220 in this case)
2. Turn off local.
3. Start playing. notice it is looping back although not set to do so in the Midi Options>Ports menu

**Expected behavior**
Don't loop back if not configured to do so.

**Setup:**
 - SD-image: 2019-12-30-zynthianos-buster-lite-1.0.0
 - Hardware Config
        Raspberry Pi 4 Model B Rev 1.1
        Soundcard: HifiBerry DAC+ ADC PRO
        Display: ZynScreen 3.5 (v1)
        Wiring: MCP23017_ZynScreen
         GPIO Expander: MCP23017
 - Software Version
        zyncoder: master (8b7bfbd)
        zynthian-ui: master (0e8f5be)
        zynthian-sys: master (1739cc2)
        zynthian-data: master (ba48e1b)
        zynthian-webconf: master (9403cc0)

jofemodo, 2020-03-10:
OK! I think i hunted the bug.
Please, update and test again!

Mickinator, 2020-03-10:
> <mailto:notifications@github.com> >
> though (#211)
> <mailto:zynthian-ui@noreply.github.com> >
> <mailto:subscribed@noreply.github.com> >

Hi Jose and Bruce,

my doubble trigger problem seems to be solved too. It would be wonderful!

yours
Micki

Gesendet mit der Telekom Mail App
<https://www.telekom.de/unterwegs/apps-und-dienste/kommunikation/telekom-e-mail?wt_mc=alias_9998_unterwegs/apps-und-dienste/telekom-email#mail-app>

closed 2020-03-10 by jofemodo

1st-BrainStormer, 2020-03-11:
Yep , that did it!   Thanks!

commit dd7611ae (2020-03-10): Fix #211: MIDI Looping back on usb devices. Fix MIDI autoconnector.