zynthian/zynthian-ui #212: presets / screens attribute missing for all JV engines

mheidt, 2020-03-12:
When selecting OBXD as engine in presets:

Nada and

Mar 12 08:36:36 zynthian zynthian_webconf.sh[3675]: ERROR:presets_config_handler: Can't initialize engine: 'NoneType' object has no attribute 'screens'
Mar 12 08:36:36 zynthian zynthian_webconf.sh[3675]: ERROR:presets_config_handler: BANK NODE 0 => 'NoneType' object has no attribute 'get_bank_list

mheidt, 2020-03-12:
Thanks.
A little more.
Renaming a bank gives a
Can't rename bank: Format doesn't match!

mheidt, 2020-03-12:
And when I try to upload KVR_Keys... i can't change presets.
It is loaded, but when I  select different presets, only the one loaded first is taken.

jofemodo, 2020-03-12:
The rename issue is solved. The KVR_keys presets loading problem seems to be specific to this bank. Probably related with "strange" characters or something like that. Let me do my research ...

mheidt, 2020-03-12:
Ah, didnt we have an issue with length?
It  should work, when I rename to a shorter bank name?

mheidt, 2020-03-12:
But renaming to KVR_Keys didn't help

jofemodo, 2020-03-12:
Seems to be a Jalv problem. Long URI's doesn't work ...

mheidt, 2020-03-13:
Ok, I found it.
That KVR_Keys has multiple Default.ttl in the manifest.
So I think, the name needs to be unique.
Maybe we should handle this in the preset2lv2?
Could you do that? @jofemodo 
I saw, that you moved the git. And actually I forgot how to test it...how the setup.py works and the like.

mheidt, 2020-03-13:
When importing with a short name, the KVR_Keys works. Renaming afterwards is not enough.
We might think about changing the uri when the bankname is shortened as well.

commit 0d530c01 (2020-03-12): Fix #212: ZynAPI instance fails because zyngui=None.
closed 2020-03-13 by mheidt