zynthian/zynthian-ui #213: RFE: Integrate pyradio as an engine

Baggypants, 2020-03-31:
**Is your feature request related to a problem? Please describe.**
OP-1 has an FM radio and we don't? Someone might like to play along to their radio? I have my zynthian next to me in the office and it's plugged in to the speakers anyway so it would nice to use it as a radio? It can also do playlists which might be good for gig accompianiment? I don't have a good use case really. 

**Describe the solution you'd like**
Implement pyradio [player.py](https://github.com/coderholic/pyradio/blob/master/pyradio/player.py) as an engine that can be assigned to a layer. Instead of choosing a patch you'd be choosing a radio station from pyradios `stations.csv`. controls for start, stop, volume up & down and mute. Add MIDI control?

**Describe alternatives you've considered**
I've not really considered this one properly, let alone an alternative :laughing: 

**Additional context**
I'm probably going to have a go at this using one of the cli engines as a template.

jofemodo, 2020-03-31:
You have been assigned for implementing this feature ;-)

riban-bw, 2024-03-12:
There is an Internet radio player in oram.

closed 2024-03-12 by riban-bw