zynthian/zynthian-ui #214: feature request for effect / layer options menu: "replace effect"

hal80, 2020-04-02:
Hi, 
when I add an effect to a layer, I have currenty the options to 1) remove all effects, and 2) Add effect.
wouldn't it make sens to also have an option to 3) replace effect?
This would apply to two user actions:
1) The bold press option menu of a syth layer, as long as there is only one effect assigned,
2) The bold press option menu of the effect itself.
The underlying idea is that one oftentimes might try out how different effects sound like and this idea with the "replace effect" option would be kind of a short cut for the user action of removing effect + adding new effect.
regards.

jofemodo, 2020-04-02:
Wish granted! ;-) Please, update and test ...

closed 2020-04-02 by jofemodo

hal80, 2020-04-02:
Cool! And you also implemented "move upchain" feature :+1:. Will test it right away.

commit 827d2191 (2020-04-02): Fix #214: Replace layer in list. Adding & delete break things.