zynthian/zynthian-ui #215: Can no longer select lv2 patches in patch grroup 'none'

Baggypants, 2020-04-02:
**Describe the bug**
When choosing patches for oxbd and padthv1 I can no longer select patches that are in the group 'None'

**To Reproduce**
Steps to reproduce the behavior:
1.  add layer obxd or padthv1
2. Choose midi channel
3. Scroll down to group 'None'
4. The cursor will stop at the group above

**Expected behavior**
Scroll to patch group 'None' and allow click

**Setup:**
 - SD-image:  Buster
 - Hardware Config: v2
 - Software Version: 

 zyncoder: master (8759b11)
zynthian-ui: master (527236f)
zynthian-sys: master (e3a2250)
zynthian-data: master (9cccef5)
zynthian-webconf: master (d72a695)

jofemodo, 2020-04-02:
OK! It's solved! Please, update and test ...

closed 2020-04-02 by jofemodo

Baggypants, 2020-04-02:
Test successful!

commit 51b4e55c (2020-04-02): Fix #215: Allow to select "None" bank on Jalv engine.