zynthian/zynthian-ui #216: scrolling long lists

hal80, 2020-04-03:
Can long lists be scrolled more efficiently?
For example, the effects layer list can have over 400 entries. Scrolling through that list is done with the _select_ knob. While in that list, the rotation function of the other knobs is unassigned - so I was thinking of the idea to assign a "quick scroll" function to one of them for browsing through that list pagewise, similar to the PageUp / PageDown buttons on a conventional computer keyboard. _Learn/Snap_ knob would be well suited for this, because it is close to the _select_ knob, which makes a one-hand operation possible. But I guess that feature is not that straight forward to implement, as the page height depends on the font size and font used.

riban-bw, 2020-04-03:
+1 for this feature. A (hopefully) simple solution might be too display each page as you scroll, i.e.

- bold press encoder 4
- rotate encoder 4
- each page displays as it is highlighted
- press encoder 4 to allow editing
This is not optimal but may be an achievable easy win.

jofemodo, 2020-04-04:
I have modified list selector for using the zyncoder acceleration feature when navigating lists of more than 96 elements. I could modify this limit easily. Modifying the acceleration params is a little bit more complex because it's hardcoded in the zyncoder C library.

Simply update and test ....

hal80, 2020-04-04:
Tested, works! :-) The acceleration-aided scrolling feature integrates really nicely and seamlessly in the user experience.

hal80, 2020-04-26:
With the latest update (around 26 april 2020), the encoder behaves strange at slow rotation in the long list of effects:
Sometimes, a change in direction of rotation is not captured from the very first step but only from the second step onwards. Sometimes, the selected line jumps one step in the wrong direction and then back and only then it follows to direction of rotation.
The encoder itself is working. There are no such jumps in other menus with less entries (i.e. with no acceleration feature).

gr0k-net, 2021-01-31:
I came here to post something similar.

As has been noted above, on many of the list pages (synths / effects / banks / presets) three out of four rotary controllers don't do anything. 

Mapping them to scroll the list like the select rotary controller may help unnecessary wear on the select controller, and for right handed people may be useful if you're for scrolling through longs lists whilst tending to more finicky device with your dominant hand.

I'm not sure if it's worth opening up a separate issue, but I've also noticed some inconsistencies with regards to favouriting an item. This works with presets, but not synths or effects. 

My list of synths is kept quite short, but especially whilst I'm new to the Zynthian and exploring what's available, my effects list is huge. Being able to favourite items would help a lot!

riban-bw, 2022-11-08:
Encoder acceleration is now implemented with improved (or at least, changed) dynamics.

closed 2022-11-08 by riban-bw