zynthian/zynthian-ui #218: RFE: Ability to overwrite a ZS3 program from the UI

jypma, 2020-04-04:
I have a little stomp-pedal with 2 buttons that can send Midi Program Change messages. It'll increment its internal counter up/down and send a message when I press one of the two buttons. It's a great way to cycle between ZS3 presets.

However, when I make adjustments, it's really hard to _overwrite_ (save) the current MIDI program (ZS3 Preset). I can only save ZS3 presets when a Midi Program Change message arrives, but the pedal will only send messages for the next or previous program; not the current one.

I can of course reprogram the pedal, but I think an "Update current ZS3 program" UI option could make sense. I can suggest two options:

a) Replace the Bold-press of _select_ in the ZS3 selection page to write to the preset instead. Make Long-press delete the preset. (I think that might be a little risky, UX-wise)
b) Replace the Bold-press of _select_ in the ZS3 selection page, to open a menu instead of "just" deleting the preset. The menu would then allow you to save to, or delete, that ZS3 preset.

Or perhaps overload one of the other buttons bold-pressed?

My current knowledge of the code is such that I could do a PR for (a), but not for (b) :-)

mheidt, 2020-04-04:
Deleting a zs3 was my request.
Would it help if I provide something in the webconf?
I added displaying the zs3 data lately.
Manually chaning the program change value or the like?

jofemodo, 2020-04-05:
I will implement the option 2 ... ;-)

jofemodo, 2020-04-05:
Done! Please, update and test ...

jypma, 2020-04-05:
Looks fine to me!

closed 2020-04-05 by jypma