zynthian/zynthian-ui #220: Controls don't always update when switching ZS3 presets on FX chain

jypma, 2020-04-05:
I have an FX chain with 5 effects layered (see attached below). I have created 2 ZS3 sub-presets, which change parameters in the "Cabinet" effect.

When I send Program Change messages (PC1 and PC2), the LV2 plugins update fine, and the sound changes. However, the on-screen controls don't always update. Sometimes they do, sometimes they don't. Sometimes they do if I send PC2 two times. 

It seems to be a race condition on the control layer itself only; a bit of debugging showed that everything from picking up the values from the ZS3 preset, to sending that down into `jalv`, works fine. Somehow the UI just doesn't pick up the change.

It's not "just" a redraw issue either: when the bug occurs, the knob "really" is at the old position, as turning it will not make a sudden jump.

jypma, 2020-04-05:
Preset to test out with:
[preset.zip](https://github.com/zynthian/zynthian-ui/files/4433240/preset.zip)

(this was saved while on https://github.com/zynthian/zynthian-ui/pull/217, in order to be able to use all plugin names, so it _may_ not load when on master branch)

jofemodo, 2020-04-10:
Hi @jypma!
I can't reproduce your problem. Your snapshot works perfectly on my zynthian, including ZS3.
Could you confirm that your software is updated to the last and you can reproduce the problem?

Thanks!

jofemodo, 2020-05-19:
Please, could you confirm the issue?

jypma, 2020-05-19:
Sorry, I missed your last message. Sure, I'll do a full upgrade and re-test. Great to hear it works for you!

jypma, 2020-05-19:
I did a full zynthian update and `apt-get upgrade`. The original preset I uploaded switches ZS3 presets fine now! Switching ZS3 presets there takes about 1 second. Thanks a lot!

However, a more complex preset I've been working on ([preset2.zip](https://github.com/zynthian/zynthian-ui/files/4649444/preset2.zip)) still switches unreliably. I've observed the following:
- Applying a preset takes ~3 seconds (raspPi 3 B)
- Zynthian doesn't apply the ZS3 preset for an incoming PC message until the _second_ time it is sent. So:
  -  I'm on ZS3-1
  - A PC2 message is received
  - The Blue MIDI indicator lights up
  - Zynthian is busy for ~3 seconds, seemingly applying a preset, but no controls update on the screen.
  - A PC2 message is received _again_.
  - The Blue MIDI indicator lights up.
  - Zynthian is busy for ~3 seconds, and now the controls _do_ update on the screen.

The above behaviour is consistent when switching any ZS3 in the attached preset. It occurs both when the repeated PC message are sent 4 seconds apart (giving Zynthian time to update), or when they're sent in rapid succession (which will make Zynthian process them both, and eventually update its on-screen controls).

I've double-checked the transmitted UDP MIDI packets with Ardour & QMidiNet, to make sure Zynthian is receiving what I think it is.

Additionally, I've observed the following:
- If I send many PC messages in succession, they're all buffered and "applied" one by one. It might be beneficial to discard queued PC messages for the same midi channel, and only apply the latest (this can probably be a different ticket).

jypma, 2020-06-08:
I think I may have nailed it down, see https://github.com/zynthian/zynthian-ui/pull/250/commits/4b4334bf70db8ed6891fb8b1f28ef01b851deddf .

jofemodo, 2020-06-10:
Woaooo! It seems you catched the bug ...
I will merge it ASAP ...

closed 2020-06-13 by jofemodo

jofemodo, 2020-06-14:
Merged and closing ...

commit 4b4334bf (2020-06-09): Update GUI when restoring a snapshot

Without this, sometimes the GUI won't update when switching between
complex ZS3 presets.

Fixes #220.
commit fce96c11 (2020-06-10): Update GUI when restoring a snapshot

Without this, sometimes the GUI won't update when switching between
complex ZS3 presets.

Fixes #220.
commit e1ed506b (2020-06-12): Update GUI when restoring a snapshot

Without this, sometimes the GUI won't update when switching between
complex ZS3 presets.

Fixes #220.
commit ed3882b2 (2020-06-13): Update GUI when restoring a snapshot

Without this, sometimes the GUI won't update when switching between
complex ZS3 presets.

Fixes #220.