zynthian/zynthian-ui #224: Settings change when stepping back from engine parameter adjust screen

riban-bw, 2020-04-10:
**Describe the bug**
After adjusting settings in an engine's parameter screen, pressing BACK will reset the engine to a preset configuration.

**To Reproduce**
Steps to reproduce the behaviour:
1. Add JV/amsynth engine
2. Select preset, e.g. BriansBank01/000: basic 11
3. Navigate to parameters screen and change some values, e.g. Master Volume = 0.5
4. Save as Snapshot
5. Restore Snapshot
6. Check volume is as previously set
7. Press BACK button to return to preset list

**Expected behaviour**
No change in configuration

**Actual behaviour**
Engine parameters are reset to default.

**Setup:**
Raspberry Pi 4 Model B Rev 1.2
Soundcard: HifiBerry DAC+ light
Display: WaveShare 7 HDMI+GPIO 1024x600
Wiring: I2C_HWC
GPIO Expander: Not detected

Raspbian GNU/Linux 10 (buster)
Build Date: 2020-03-28

zyncoder: master (11e32fa)
zynthian-ui: master (477ee17)
zynthian-sys: master (a305ab9)
zynthian-data: master (9cccef5)
zynthian-webconf: master (ab294aa)

jofemodo, 2020-04-10:
It should be fixed now ... although it's not perfect.
Preset preload/restore should be implemented as a kind of ZS3 ...

closed 2020-04-10 by jofemodo

riban-bw, 2020-04-10:
Great, thanks! That has been bugging me for ages but only just had time to figure out what was going wrong.

commit 6f4c2646 (2020-04-10): Fix #224: Fix preload code for avoiding unneeded preset loading.
Restoring doesn't restore controller values. ZS3s could be used for solving that.