zynthian/zynthian-ui #225: error stoping midi play

le51, 2020-04-13:
I'm feeding zythian with midi sent from a step sequencer.
When I push sequencer's stop button (midi log: "SYS STOP"), UI goes black and the following error message is shown:

"error stoping midi play"
" 'noneType' object has no attribute 'pid' "

riban-bw, 2020-04-14:
This is because there is a mapping hardcoded for:
MIDI | CUIA 
-|-
START | START_MIDI_PLAY 
CONTINUE | START_MIDI_PLAY  
STOP | STOP_MIDI_PLAY

This in turn is hardcoded to trigger the MIDI recorder's start / stop playing. If a MIDI track is not selected then this fails. I have coincidentally found the same issue whilst building a step sequencer for Zynthian. I am about to propose changing the behaviour of this to allow configuration of MIDI timecode and transport messages.

jofemodo, 2020-04-14:
It's solved. Please, update and test.

closed 2020-04-14 by jofemodo

le51, 2020-04-14:
It's okay ! Thanks.

commit 3c865843 (2020-04-14): Fix #225,#226: Fix error when calling stop recorder CUIA and no recording has been started.