zynthian/zynthian-ui #229: Program Change does not work as it should

Mickinator, 2020-04-14:
If I send a Program Change command Zynthian does't reakt in all situations

*To Reproduce**
My Organs are in Folder 001.
Nr 6 is Jimmy Smith, Nr 7 is John Lord.
If I send MSB H BF 00 01, LSB H BF 20 01, PC H CF 07 it works all well.
If I send MSB H BF 00 01, LSB H BF 20 01, PC H CF 06 Zynthian does not finnish the call. No sound is to hear.

My Midi Options are ser to Master Midi Channel = 16, Master Change Type = Custom, Master Bank Change Mode = MSB.

If tried to send other Combinations of MSB, LSB, PC, but I had no good results. The only stable version to call John Lord is the showed above 1 1 7.

[image]

[image]

This foto shows, where Zynthian stops
[image]

This is a Midi-Ox-Foto.  The first PC doesn work, the secound does.
[image]

jofemodo, 2020-04-14:
Can you confirm that your snapshots are working OK when loaded from the UI?

Mickinator, 2020-04-14:
> <mailto:notifications@github.com> >
> should (#229)
> <mailto:zynthian-ui@noreply.github.com> >
> <mailto:norbert.schreiber@t-online.de> >, Author <
> author@noreply.github.com <mailto:author@noreply.github.com> >

No, i can't. I was a while only working with the 'John Lord' spapshot and 
changed the snapshots only via midi. Now I see, it's a question of every 
program change.
Via UI and via Midi 1 1 7 is working, 1 1 6 not.

Tomorrow i will send you some snapshots.

Best thanks for your work!

Gesendet mit der Telekom Mail App
<https://www.telekom.de/unterwegs/apps-und-dienste/kommunikation/telekom-e-mail?wt_mc=alias_9998_unterwegs/apps-und-dienste/telekom-email#mail-app>

jofemodo, 2020-11-27:
Hi @Mickinator !

Did you advance with this issue? Is the problem still there with latest SD images?

Mickinator, 2020-11-28:
Hi Jose,
please excuse. Momentary I´ve a lot of trouble with my 85 years old aunt. She is sick with demencial problems and I forgot this issue.

I´ve made a lot of updates in the last time and at one point the prob was solved. 
Please set this issue to solved.

I´ve tried to install the new image to a sd, but after burning my PC says, there is an incorrect checksum and the sd does not work. 
Because the probs with my aunt i didn´t take newer tries.

Keep healthy and thanks for your work !

jofemodo, 2020-11-28:
Latest official SD image (Buster RC-2) should run like a charm, and also some of the latest nightly builds.

closed 2020-11-28 by jofemodo