zynthian/zynthian-ui #235: Change Midi Channel of first layer

hal80, 2020-04-16:
If I have just one single synth layer, it's possible to clone it to another midi channel, but there is no option to change the assigned midi channel.
Only if there are more than one layer, it's possible to change their respective midi channel.
Is there a certain reason why changing midi channel is not implemented if there is only one layer?
Could it be implemented?

jofemodo, 2020-04-16:
I can't reproduce your problem. It works perfectly for me.
What engine are you using for the first layer? Please, remember to specify the steps for reproducing an error when opening an issue. It saves time ;-)

hal80, 2020-04-17:
How could I forget ;-)
Sorry, in fact it's not the fist layer, but it's the Pianoteq Engine in any layer which has no "change midi channel" option. This is reproducable at any state / configuration the zynthian is in, no matter what other layers exist and what midi channels are used. Maybe its due to licensing, because it can only run in once instance.
(I had initially assumed that the order in which the layers are created plays a role, but actually I was loading Pianoteq as the first layer for most of the time.)
Im running https://os.zynthian.org/2019-09-24-zynthianos-buster-lite-1.0.0-RC1.zip which I have regularly updated.
zyncoder: master (3513de2)
zynthian-ui: master (926af8e)
zynthian-sys: master (ba3d180)
zynthian-data: master (9cccef5)
zynthian-webconf: master (ab294aa)

jofemodo, 2020-04-17:
Sorry, currently Pianoteq layers can't be moved to another MIDI channel because the engine have to be started in fixed layer. I will try to study how to improve this ... perhaps making Pianoteq to listen all layers (i think this is possible) and filtering on zynthian.

Regards,

closed 2020-05-03 by jofemodo

jofemodo, 2020-05-07:
Hi @hal80 ,

Could you confirm that it works for you?

Thanks!

hal80, 2020-05-07:
Yes it works! I can change it to other channels and back, and also if there are other midi channels occupied by other layers it works. Thanks for implementation :)

commit 77bf0b86 (2020-05-03): Fix #235: Allow changing MIDI channel on layer options for Pianoteq.