zynthian/zynthian-ui #236: Add step sequencer

riban-bw, 2020-04-16:
It would be advantageous to add a step sequencer to Zynthian.

**Requirements**

|ID|Requirement|Status|
|-|-|-|
|M1|A new page accessible from "Main" page|Done|
|M2|Pattern editor|Done|
|M3|Song editor|Proposed|

**Pattern Editor**
|ID|Requirement|Status|
|-|-|-|
|P1|Grid showing notes with start and duration indicated horizontally and note number indicated vertically|Done|
|P2|All step columns displayed|Done|
|P3|Zoom quantity of note rows displayed with encoder|Done|
|P4|Cursor / cell highlight indicates selected note|Done|
|P5|Move cursor around grid using two rotary encoders|Done|
|P6|View scrolls when limits reached|Done|
|P7|View can be scrolled by click / touch and drag|Done|
|P8|Add and remove notes by pressing encoder button|Done|
|P9|Add and remove notes by touching / clicking on grid|Done|
|P10|Note velocity can be adjusted by adjusting an encoder|Done|
|P11|Note duration can be adjusted by adjusting an encoder|Proposed|
|P12|Playback can be started and stopped by clicking an encoder|Done|
|P13|Playback can be started and stopped by touching / clicking the screen|Proposed|
|P14|Playback can be started and stopped by sending MIDI messages|Done|
|P15|Playback can sync to MIDI timecode or internal clock|Proposed|
|P16|Tempo of internal clock can varied by adjusting an encoder|Proposed|
|P17|Playback at various divisors, e.g. /2, /4, /8, etc.|Proposed|
|P18|Playback can be set to forward, reverse, bounce|Done|
|P18.1|Playback can be set to random|Proposed|
|P19|Patterns can be copied|Done|
|P20|Patterns can be deleted|Done|
|P21|Quantity of step in each pattern can be adjusted from 2 - 64|Done|
|P22|MIDI output can be sent to any MIDI channel|Done|
|P23|Input can be set by external MIDI keyboard during loop (record mode)|Proposed|
|P24|Option to show vertical grid lines at configurable intervals, e.g. beats / bars|Done|
|P25|Option to constrain available notes, e.g. to pentonic scale, drum map, etc.|Proposed|
|P26|Option to transpose whole pattern|Done|
|P27|Option to change to display time vertically and notes horizontally|Proposed|
|P28|Use zynthian display configuration for colours and fonts|Done|
|P29|Integrate zynthian status bar|Proposed|
|P30|Continuous Controller editor|Proposed|

**Song editor**
|ID|Requirement|Status|
|-|-|-|
|S1|Grid showing channels vertically and time horizontally|Proposed|
|S2|All channel rows displayed|Proposed|
|S3|Each cell in grid may contain a pattern|Proposed|
|S4|Patterns can be any duration, i.e. not necessarily vertically aligned|Proposed|
|S5|Patterns may be positioned at any time, not just grid boundaries|Proposed|
|S6|Cursor / cell highlight indicates selected pattern|Proposed|
|S7|Move cursor around grid using two rotary encoders|Proposed|
|S8|View scrolls when limits reached|Proposed|
|S9|View can be scrolled by click / touch and drag|Proposed|
|S10|Add and remove patterns by pressing encoder button|Proposed|
|S11|Add and remove patterns by touching / clicking on grid|Proposed|
|S12|Coarse navigation using encoder|Proposed|
|S13|Fine navigation using encoder whilst button pressed|Proposed|
|S14|Tracks can be copied|Proposed|
|S15|Songs can be copied|Proposed|
|S16|Songs can be deleted|Proposed|
|S17|Option to transpose whole song|Proposed|
|S18|Import from .mid file, automatically creating patterns|Proposed|

**Describe alternatives you've considered**
LV2 plugins and other existing step sequencers. These don't currently integrate with Zynthian UI. External hardware sampler. This wouldn't give us a single box solution.

**Additional context**
Work has begun on prototype by @riban-bw.

riban-bw, 2020-04-18:
> Driving the Interface from a mouse doesn't seem to have an exit.
> Suggestion similar back button at top of screen as defined for other zynthian screens.

Fixed in latest commit.

le51, 2020-05-02:
Hi Riban,
I've just upgraded zynthian using the stepseq branch (latest commit 6faf029). 

- The build process (to build the .so lib) is not done during update. I have had to do it by hand.
- layer encoder allow me to navigate trough the different items menu but once an item is selected (saying tempo) I can't make any change.
I saw that you have published a couple of commits in the last few hours, so maybe you know that already and it's in your todo list.
Good luck !

riban-bw, 2020-05-02:
I have added the library binary to github and requested guidance from @jofemodo as to whether to build from source or distribute binary in future release. The layer encoder is not working in master branch. I need to review the feedback from @jofemodo on a patch I provided to figure out how to resolve this.

riban-bw, 2020-05-03:
Latest commit to feature/stepseq branch fixes layer encoder issue.

hal80, 2020-05-03:
Another feature request: monophonic mode for monophonic synthesizers which would be really easy to implement: restrict to one single note at a time in the pattern editor. Entering a new note in the same step deletes the old note at the same time. This would be really fabulous and would allow a very direct interaction with e.g. externally connected synths (I have one here on my table :))

jofemodo, 2020-05-04:
I just modified the update script for building the zynseq library when needed.

riban-bw, 2020-05-04:
@hal80 I added your monophonic mode feature request as feature P31. We discussed this during Zynth Club tonight and I am not sure this is a good idea. There would be many rulesets to agree like whether adding a new, overlapping note is prohibited, deletes existing notes or truncates them? Also how do you do legato? I would suggest that the polyphonic step sequencer with human control may prove a more elegant solution than having lots of switches to change modes. We could (possibly) add the ability to highlight overlapping notes. This is not quite as trivial as it first seems because you need to check all 127 notes for starts or ends that fall within any other note's start or end. I will leave it on the feature list for now.

hal80, 2020-05-05:
@riban-bw When I thought about it, I also realized that it is actually not as trivial as it seems at first glance. What I had in mind was a very simple mode for bass lines with notes that stick strictly to the grid. Maybe quarter and eighth notes. I would argue for keeping the request on the list and if there is a chance to start with such a simple mode and we could see how it works in practice. How to handle overlapping notes, I would say that everything that even partially overlaps is deleted when new notes are entered. For legato notes I've got no idea at the moment. Is it different from poly mode?

jofemodo, 2020-05-11:
Hi @riban-bw!

This is "more or less" a copy from my discourse's comment:

https://discourse.zynthian.org/t/step-sequencer/4043/237

**Issue list / Things To Improve:**

**1.)** Loop / stop button on topbar.
When enabled with touch, it changes color. When using the switch, color doesn't change.

**2.)** Encoder assignment changed:

> 
> * LAYER: Note duration
> * SNAPSHOT: Note velocity
> * BACK: Note selection
> * SELECT: Step selection / menu selection / parameter value
> 

I really love the duration/velocity interaction, but regarding velocity, i would like:

-  having numerical feedback apart from color change. For duration i feel current graphic feedback is enough. (i know this is in the current issue list, but i wanted to add my POW) 
- use encoder acceleration. It takes too much to change values in the 0-127 range ...

**3.)** I don't like the "back" entry on the menu, but i don't have a better proposal yet ... :yum: In fact, mixing in the same menu parameters like "steps per beat" and a link to "Song Editor" ...  OK, i know you can feel it too ... :grin: 

**4.)** Pad Trigger => I think "Launch Pad" is a better name, and much more "popular".

**5.)** Refreshing grid seems to be slow. I don't know if speed can be improved, but if it's not easy, you could think of refreshing only when parameter is asserted.

**6.)** Editor appears in topbar to allow change of a parameter selected by menu.
I suppose you mean the 4 arrow-buttons. It seems to be only useful for touch-screen users, being confuse and distracting for the rest. I would like to have the option of enabling/disabling this kind of touch-only widgets, so UI experience can be optimized. Encoder interface users normally have smaller displays, so screen space is precious, etc.

**7.)**
[quote="riban, post:158, topic:4043"]
This version is still using libseq internal clock.
[/quote]
This is the main stopper for merging with master. As soon as the stepseq be a pure clock-consumer, i will start merging with master.

Regarding development & debugging itself, please, please, remove the:

- I just disabled the "Unhandled MIDI message XXX", as it was polluting the UI log, making very difficult to use it for debugging purposes. Don't forget to pull the change to your local branch.

jofemodo, 2020-05-12:
**Today's Zynstep Update Report:**

First the goods:

+ The velocity level is much more visible now. For me it's OK like that.

+ Encoder acceleration seems to work. It's OK for parameters like velocity, but  would prefer normal behaviur (non-accelerated) when changing duration or moving on the grid, specially for X axis (step). 

+ After removing binary, libseq seems to build OK.

And now the issue report:

+ When starting the UI, the screen goes to white for some seconds, what it's not very nice. This doesn't happens on master, so probably it's the zynthian_gui_stepsequencer (or some of the new screens) ...

+ I can confirm the issue with the step 0. It can't be reached with encoders. It seems a problem with zyncoder setup.

+ When changing "steps in pattern", notes are deleted when reducing number of steps, although finally decide to not assert the change, or choose a bigger number.

+ When increasing steps per beat, zynthian UI get frozen and have to restart. This is the log dump:
```
May 12 23:15:23 zynthian startx[28682]: ERROR:zynthian_gui: list index out of range
May 12 23:15:23 zynthian startx[28682]: Traceback (most recent call last):
May 12 23:15:23 zynthian startx[28682]:   File "./zynthian_gui.py", line 1143, in zyncoder_read
May 12 23:15:23 zynthian startx[28682]:   self.screens[self.modal_screen].zyncoder_read()
May 12 23:15:23 zynthian startx[28682]:   File "/home/pi/zynthian-ui/zyngui/zynthian_gui_stepsequencer.py", line 748, in zyncoder_read
May 12 23:15:23 zynthian startx[28682]:     self.zyncoderOwner[encoder].onZyncoder(encoder, value - 64)
May 12 23:15:23 zynthian startx[28682]:   File "/home/pi/zynthian-ui/zyngui/zynthian_gui_stepsequencer.py", line 734, in onZyncoder
May 12 23:15:23 zynthian startx[28682]:     self.changeParam(value)
May 12 23:15:23 zynthian startx[28682]:   File "/home/pi/zynthian-ui/zyngui/zynthian_gui_stepsequencer.py", line 669, in changeParam
May 12 23:15:23 zynthian startx[28682]:     result=self.paramEditor['onChange'](self.paramEditor['value'])
May 12 23:15:23 zynthian startx[28682]:   File "/home/pi/zynthian-ui/zyngui/zynthian_gui_patterneditor.py", line 505, in onMenuChange
May 12 23:15:23 zynthian startx[28682]:     stepsPerBeat = STEPS_PER_BEAT[value]
May 12 23:15:23 zynthian startx[28682]: IndexError: list index out of range
```

Cheers!

riban-bw, 2020-05-13:
Issue with white screen seems to be related to snapshot loading. If instantiation of zynthian_gui_stepsequencer occurs after `if zynthian_gui_config.restore_last_state:` block within zynthian_gui.py then the delay is not evident (maybe masked). I am not sure what the issue is. If @jofemodo has any insight I would appreciate your input.

It will be good to get this resolved so we can merge this with master branch which will facilitate the reporting of individual bugs as separate github tickets rather than append to this feature request.

jhnnhgh, 2020-05-21:
Firstly, awesome work so far. 

I would like to propose an endless sequencer feature like the OP-1 has (https://teenage.engineering/guides/op-1/sequencers). This would be a great adition to the workflow, but I'm not sure how viable it is.

gr0k-net, 2021-02-11:
Not sure if this is still the best place to leave feedback on the step sequencer.

One thing I've noticed is that the pattern editor view defaults to middle C, which is fine on an empty pattern, but rather annoying when working on bass lines or with samples mapped across a keyboard. 

Perhaps when loading a pre-existing pattern the sequencer takes the highest note, and the lowest note, and centers the screen in the middle of the two? OR alternatively, looks for the highest density of notes, and centers the screen around that?

riban-bw, 2021-02-11:
@gr0k-net that is a very good point. I would find this useful when editing drum patterns that tend to use lower range. There may be some coplexity in the rules that decide when to scroll the display, e.g. if you access Pattern Editor and manually scroll to the area of interest then leave and return to Pattern Editor you may wish to return to the last scrolled location rather than the calculated _most likely_ location. Of course this is the same issue as we have now with the default being middle C but if we adjust this it may be worth considering if there are rules that can help. Also this may result in apparently inconsistent behaviour with automatic scrolling taking you to a different place depending on the content of the pattern, e.g. first time may take you to B0 because you have a bass drum and next time may take you to A4 because you now have a Triangle. Please consider these points and offer suggestions. I have added this to the low priority list of tasks.

jofemodo, 2021-02-11:
Hi @riban-bw !

I also find this quite annoying and think than taking the middle note between the highest and lower can be a good starting point. If not, remembering the latest cursor position could work OK.

Regards,

riban-bw, 2021-02-13:
I have implemented two changes for pattern editor:

- Vertical scroll position is saved for each pattern so each time you enter pattern editor you will get same location
- Entering pattern editor now selects the centre row rather than the bottom row
 
I found this more intuitive and useful than adding an algorithm to detect the highest density of notes (which would have been rather complex). You generally want to get back to where you were. If you added a bass line then you probably left the editor where your notes are.

gr0k-net, 2021-02-16:
Haven't had a heap of time to play around this last couple of days, but it seems to be working well!! Thank you!

I also noticed the new pad visuals. The loop mode is far more obvious, but active pads slightly less so. Also noticed up to 24 steps per beat now!

If I can be so rude, I have another request, semi-related to visual feedback. With regards to trigger notes on pads, could you also allow CC and/or note-on/note-off messages?

I am using a friends MPD at the moment, and previously an MPK Mini, and both have pads or controls that can be mapped to toggle via note or CC, but there's no way I've found in ZynSeq to use CC messages, and I have to either program them for momentary rather than toggle, and therefor get no visual feedback, or use note-on/note-off (127/0) toggles, but 'double click' to keep visual feedback consistent on the controller (LED pads, or manually mapped CC controls)

Also, loving the live recording, but noticed that programmed notes are related to steps per beat rather than currently selected beat length like they were previously. Ideally velocity and length of note would depend on velocity and note-on/note-on from the controller, but I imagine this is a little more complex than an either/or between current step-size or note properties.

Thank you again for everything! Please don't take my feedback and requests as lack of gratitude, I won't stop reminding you how much I'm loving the Zynthian, and it's largely because of ZynSeq. Thank you again!

riban-bw, 2021-02-16:
@gr0k-net I will look at how note length may be improved for live input. I don't think it has changed - is adds a single step duration note. I can see it would be advantageous to either use the currently configured note duration or the actual note on / note off events. The former is challenging because it is not currently present within the library, being implemented in the Python application. The latter should be possible but will be quite challenging to correlate the events over time. It might not happen in the short term!

Regarding toggle and CC: There is an issue with asynchronous toggle that states may get out of step. We could implement an option to allow sequences to be started with note on and stopped with note off but you do have the issue that exclusive groups will stop each other so your controller will show the wrong state if the sequence stops. If the controller supports feedback, lighting pads when a note on is received then you can use zynpads feedback / tally option. (I don't think the MPK or MPD supports this.) I also worry about adding complexity to the configuration. We are trying to minimise the quantity of menus and reduce complexity where possible. Maybe we can look at improving integration with pad controllers in a later release. I did buy a pad controller but it was rubbish and broken so it went back to the supplier and I don't have anything to test with now other than normal keyboard MIDI controllers.

Please remember your feature requests as you may need to resubmit after we release zynseq to the stable branch.

gr0k-net, 2021-02-18:
Thanks again @riban-bw!

I hadn't thought about the exclusion groups, they will be impossible to keep in sync. Hmmmmmm... I will do some thinking.

After playing around last night, I'm finding the play icon adequate. It was just that it took a minute to get used to that's all.

Happy to re-submit features after it reaches stable.

Thank you again!

riban-bw, 2021-02-20:
ZynSeq is now implemented. Individual feature requestes and bugs have been added to GitHub. I think it is time to retire this ticket. Goodby old friend...

closed 2021-02-20 by riban-bw