zynthian/zynthian-ui #237: Capture stderr from Pure Data patches

tstex, 2020-04-17:
This ticket references some issues I documented here: https://discourse.zynthian.org/t/pd-crashing/4045

Basically, output from a midi foot pedal, crashes PD in such a way that it takes down jackd sometimes, and the whole Zynthian UI other times. So far only a hardware reboot enables me to play again. While ultimately I would like to solve the crash issue, for starters I just want to catch stderr from my PD patch. 

I have added the -stderr flag to the startup command for pure data in zynthian_engine_puredata.py, but still aren't seeing that output anywhere. I considered adding another FileHandler to logging in the zynthian_engine.py so I could control where output went, but have since abandoned that task.

jofemodo, 2020-05-18:
You should try to add something like:

`self.proc.logfile_read = sys.stderr
`
just after zynthian_engine's line 88, on the start method.

Regards,

jofemodo, 2020-09-22:
Hi @tstex !
Did you solve this issue? Could i close it?

tstex, 2020-09-22:
I solved my issues by transferring my systems to a raspberry pi 4B. Never
get a chance to try this, but let's call it fixed.

Thanks!

closed 2020-09-22 by jofemodo